cartographer-project/cartographer_ros #15: Could you list some tutorial and some paper about this project

kintzhao, 2016-08-06:
Thank your for your open sources. Could you list  some tutorial and some paper about this project.

kintzhao, 2016-08-06:
I use the turtlebot_gazebo  simulate to launch robot and setup new cartographer_ros launch file tb.launch .
`<launch>
  

  

  <node name="cartographer" pkg="cartographer_ros"
      type="cartographer_node" output="screen" >
    <rosparam subst_value="true">
      map_frame: "map"
      odom_frame: "odom"
      tracking_frame: "base_link"
      provide_odom: true #false
      configuration_files_directories: [
        "$(find cartographer_ros)/configuration_files"
      ]
      mapping_configuration_basename: "turtlebot.lua"
      imu_topic: "/mobile_base/sensors/imu_data" #"/imu"
      laser_scan_2d_topic: "/scan"  #"/horizontal_2d_laser"
      laser_min_range_m: 0.
      laser_max_range_m: 30.
      laser_missing_echo_ray_length_m: 5.
    </rosparam>
  </node>

</launch>`

I can only read the topic of '/cartographer/submap_list'  and it  only contain the pose message, How can i display the map(or submap) using rviz. 

could you create a launch file and rosbag dataset for test.

kintzhao, 2016-08-06:
[image]

kintzhao, 2016-08-06:
My output message. 
---------------------------------------------------------------------------
roslaunch cartographer_ros tb.launch 
... logging to /home/yhzhao/.ros/log/3edb9664-5bd9-11e6-b362-08626614ffb2/roslaunch-X550JD-30278.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://X550JD:54294/

SUMMARY
========

PARAMETERS
 * /cartographer/configuration_files_directories: ['/home/yhzhao/ro...
 * /cartographer/imu_topic: /mobile_base/sens...
 * /cartographer/laser_max_range_m: 30.0
 * /cartographer/laser_min_range_m: 0.0
 * /cartographer/laser_missing_echo_ray_length_m: 5.0
 * /cartographer/laser_scan_2d_topic: /scan
 * /cartographer/map_frame: map
 * /cartographer/mapping_configuration_basename: turtlebot.lua
 * /cartographer/odom_frame: odom
 * /cartographer/provide_odom: True
 * /cartographer/tracking_frame: base_link
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /
    cartographer (cartographer_ros/cartographer_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[cartographer-1]: started with pid [30296]
[ INFO] [1470489947.565382858]: I0806 21:25:47.000000 30296 submaps.cc:175] Added submap 1
[ WARN] [1470489949.147453833, 16.970000000]: W0806 21:25:49.000000 30296 cartographer_node_main.cc:288] Cannot transform camera_depth_frame -> base_link: Lookup would require extrapolation into the past.  Requested time 15.780000000 but the earliest data is at time 16.100000000, when looking up transform from frame [camera_depth_frame] to frame [base_link]
[ WARN] [1470489950.147332892, 17.970000000]: W0806 21:25:50.000000 30296 cartographer_node_main.cc:288] Cannot transform camera_depth_frame -> base_link: Lookup would require extrapolation into the past.  Requested time 16.070000000 but the earliest data is at time 16.100000000, when looking up transform from frame [camera_depth_frame] to frame [base_link]
[ INFO] [1470489950.148906089, 17.970000000]: I0806 21:25:50.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to -nan%.
[ INFO] [1470490006.289751904, 74.080000000]: I0806 21:26:46.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 3.4%.
[ INFO] [1470490063.620978942, 131.380000000]: I0806 21:27:43.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 3%.
[ INFO] [1470490121.353868794, 189.080000000]: I0806 21:28:41.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.8%.
[ INFO] [1470490140.920690635, 208.640000000]: I0806 21:29:00.000000 30327 sparse_pose_graph.cc:258] We caught up. Hooray!
[ INFO] [1470490178.786025220, 246.480000000]: I0806 21:29:38.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.65%.
[ INFO] [1470490236.407092612, 304.080000000]: I0806 21:30:36.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.56%.
[ INFO] [1470490293.544183016, 361.180000000]: I0806 21:31:33.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.5%.
[ INFO] [1470490349.121800472, 416.730000000]: I0806 21:32:29.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.45714%.
[ INFO] [1470490364.909786798, 432.510000000]: I0806 21:32:44.000000 30296 submaps.cc:175] Added submap 2
[ INFO] [1470490375.036676295, 442.630000000]: I0806 21:32:55.000000 30328 sparse_pose_graph.cc:258] We caught up. Hooray!
[ INFO] [1470490404.704632481, 472.280000000]: I0806 21:33:24.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.425%.
[ INFO] [1470490460.734398055, 528.280000000]: I0806 21:34:20.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.4%.
[ INFO] [1470490515.713896148, 583.230000000]: I0806 21:35:15.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 2.38%.
[ WARN] [1470490550.684089444, 618.180000000]: W0806 21:35:50.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490550.832159265, 618.330000000]: W0806 21:35:50.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490550.930790741, 618.430000000]: W0806 21:35:50.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.030625031, 618.530000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.182643259, 618.680000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.281517098, 618.780000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.426809701, 618.920000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.530467060, 619.030000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.636505336, 619.130000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[ WARN] [1470490551.780728655, 619.280000000]: W0806 21:35:51.000000 30296 local_trajectory_builder.cc:178] Dropped empty horizontal laser point cloud.
[... cut ...]
[ INFO] [1470490572.345736172, 639.830000000]: I0806 21:36:12.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 3.01818%.
[ INFO] [1470490609.742265126, 677.210000000]: I0806 21:36:49.000000 30296 submaps.cc:175] Added submap 3
[ INFO] [1470490627.229128887, 694.690000000]: I0806 21:37:07.000000 30296 motion_filter.cc:42] Motion filter reduced the number of scans to 3.05%.
[ INFO] [1470490630.090986653, 697.550000000]: I0806 21:37:10.000000 30331 sparse_pose_graph.cc:258] We caught up. Hooray!

wohe, 2016-08-08:
Hi,

A tutorial and an example bag is definitely planned, but it might take us a few days.

The output you get looks good. Getting the generated map displayed in rviz should be straightforward:

1. Make sure rviz can find the cartographer_ros plugin (e.g. source the catkin setup.bash)
2. Start rviz
3. Click on "Add" to add a display
4. You should see a cartographer_ros/Submaps display type, select it
5. The Submaps display has a "Topic" property where you have to select /cartographer/submap_list

Please let me know, if any of the steps is unclear or does not work as expected.

kintzhao, 2016-08-09:
Thanks, i find it at By display type of rviz(i still find it at By topic).  then rviz can display ok, I will test it and read it.

closed 2016-08-09 by kintzhao