cartographer-project/cartographer_ros #17: The warning about tf laser frame_id , check  msg->header.frame_id

kintzhao, 2016-08-09:
Is this a bug in following function,  when i test it in default,(my /scan frame_id is "/camera_depth_frame"). 
Print the Warning:
Warning: Invalid argument "/camera_depth_frame" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like: 
         at line 130 in /tmp/binarydeb/ros-indigo-tf2-0.5.13/src/buffer_core.cpp

Only I change code and using "camera_depth_frame" is fine.  "/camera_depth_frame"  and msg->header.frame_id  all print warning .
If i not change,  rosrun  tf tf_echo /map /camera_depth_frame will  not output the transform . Change it , OK.
----------------------
void Node::LaserScanMessageCallback(
    const sensor_msgs::LaserScan::ConstPtr& msg) {
  auto sensor_data = ::cartographer::common::make_unique<SensorData>(
      "camera_depth_frame"/* "/camera_depth_frame"*//*msg->header.frame_id*/, ToCartographer(*msg));
  sensor_collator_.AddSensorData(
      kTrajectoryId,
      ::cartographer::common::ToUniversal(FromRos(msg->header.stamp)),
      laser_2d_subscriber_.getTopic(), std::move(sensor_data));
}
-----------------------------------

skohlbr, 2016-08-09:
AFAIK this is a change in tf2. In earlier times, a preceding slash in tf was ok, but this was changed in tf2 along with getting rid of tf_prefix. You can find many instances of people running into this issue and more information by googling for "in tf2 frame_ids cannot start with a '/' like". This is not a issue with cartographer_ros, but the tf2 library it uses.

wohe, 2016-08-10:
This is also my understanding, see http://wiki.ros.org/tf2/Migration. The proper fix is to remove the leading "/" from your frame_id. Seems we were giving a bad example in one place, so I sent out #19.

closed 2016-08-10 by wohe