cartographer-project/cartographer_ros #32: Tf extrapolation warnings with some setups

skohlbr, 2016-08-29:
I've been trying out the system with a simulated spinning lidar and get quite a bunch of tf related warnings when feeding bag data to cartographer_ros. I thus think there are some waitForTransform() calls missing. See console output here: [tf_warnings_spinning_lidar.txt](https://github.com/googlecartographer/cartographer_ros/files/442195/tf_warnings_spinning_lidar.txt). As LIDARs are mounted fixed in most other example setups, the missing waits likely didn't appear there due to fixed joints being forward timestamped by robot_state_publisher per default.

To reproduce:
* Download bag file: https://drive.google.com/drive/folders/0B1hU91jkd7Vwa3lXZVJPbmZUTWs (Video: https://www.youtube.com/watch?v=KIdIpFkSiHE)
* Get spinning lidar setup from https://github.com/skohlbr/cartographer_ros/commit/90bd6a771014f8a2f07a37974a3bcab3647bc6f2
* Run demo (change command below appropriatly)
```
roslaunch cartographer_ros demo_robot_spin_lidar.launch bag_filename:=/home/kohlbrecher/Downloads/2016-08-04-15-17-14.bag 
```
* Note the launch setup uses https://github.com/team-vigir/vigir_lidar_proc/tree/master/vigir_laserscan_to_pointcloud so far,, it would be preferable to get rid of this dependency.

skohlbr, 2016-08-29:
Ok, found the culprit: https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/src/cartographer_node_main.cc#L97
This is currently a constant double, any reason not to make this a parameter that can be set from outside (LUA config or ROS param)?

damonkohler, 2016-08-29:
Sure, that can be a Lua parameter. The value was chosen experimentally. Will you file a PR or shall we handle it?

skohlbr, 2016-08-29:
I'll make a PR, looks straightforward.

PR #33 (2016-08-30): Adds lua parameter for setting transform wait time
Fixes #32 by allowing user to specify tf lookup wait time via the `transform_wait_time` lua parameter. Adjusted the existing config files accordingly.
closed 2016-08-30 by damonkohler