cartographer-project/cartographer_ros #37: eigen_conversions are required.

SirVer, 2016-08-30:
When following what the docker image is doing, I get the following error:

~~~
==> Processing catkin package: 'cartographer_ros'
==> Creating build directory: 'build_isolated/cartographer_ros'
==> Building with env: '/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer/env.sh'
==> cmake /usr/local/google/home/hrapp/prog/catkin_workspace/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/install_isolated -G Unix Makefiles in '/usr/local/google/home/hrapp/prog/catkin_workspace/build_isolated/cartographer_ros'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros
-- Using CMAKE_PREFIX_PATH: /usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/ceres_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/glog_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/gflags_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/catkin_simple;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros_msgs;/opt/ros/indigo
-- This workspace overlays: /usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/ceres_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/glog_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/gflags_catkin;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/catkin_simple;/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros_msgs;/opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /usr/local/google/home/hrapp/prog/catkin_workspace/build_isolated/cartographer_ros/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.18
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "eigen_conversions"
  with any of the following names:

    eigen_conversionsConfig.cmake
    eigen_conversions-config.cmake

  Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
  set "eigen_conversions_DIR" to a directory containing one of the above
  files.  If "eigen_conversions" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:33 (find_package)

-- Could not find the required component 'eigen_conversions'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "eigen_conversions"
  with any of the following names:

    eigen_conversionsConfig.cmake
    eigen_conversions-config.cmake

  Add the installation prefix of "eigen_conversions" to CMAKE_PREFIX_PATH or
  set "eigen_conversions_DIR" to a directory containing one of the above
  files.  If "eigen_conversions" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:33 (find_package)

-- Configuring incomplete, errors occurred!
See also "/usr/local/google/home/hrapp/prog/catkin_workspace/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/usr/local/google/home/hrapp/prog/catkin_workspace/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer/env.sh', 'cmake', '/usr/local/google/home/hrapp/prog/catkin_workspace/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /usr/local/google/home/hrapp/prog/catkin_workspace/build_isolated/cartographer_ros && /usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer/env.sh cmake /usr/local/google/home/hrapp/prog/catkin_workspace/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/usr/local/google/home/hrapp/prog/catkin_workspace/install_isolated -G 'Unix Makefiles'

Command failed, exiting.

~~~

damonkohler, 2016-09-02:
Not reproducible.

closed 2016-09-02 by damonkohler

Kailegh, 2016-11-09:
it happens the same to me, have you been able to solve it?

SirVer, 2016-11-09:
@Kailegh try installing `ros-indigo-eigen-conversions` or equivalent for your ROS distro.

Kailegh, 2016-11-09:
i tried that and still does not work

I tried this in order to solve it:

https://github.com/ros/cmake_modules/blob/0.3-devel/README.md#usage

and now i get this error 
```
In file included from /home/lsi2/catkin_ws2/src/cartographer/cartographer/io/points_batch.cc:17:0:
/home/lsi2/catkin_ws2/src/cartographer/cartographer/io/points_batch.h:23:22: fatal error: Eigen/Core: No existe el archivo o el directorio
 #include "Eigen/Core"
                      ^
compilation terminated.
[ 30%] Running Sphinx v1.2.2
Building CXX object cartographer/mapping/CMakeFiles/mapping_probability_values.dir/probability_values.cc.o
make[2]: *** [cartographer/io/CMakeFiles/io_points_batch.dir/points_batch.cc.o] Error 1
make[1]: *** [cartographer/io/CMakeFiles/io_points_batch.dir/all] Error 2
make[1]: *** Se espera a que terminen otras tareas....
loading pickled environment... done
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.
[ 30%] Built target build_doc
In file included from /home/lsi2/catkin_ws2/src/cartographer/cartographer/common/math.h:20:0,
                 from /usr/include/c++/4.8/cmath:44,
                 from /home/lsi2/catkin_ws2/src/cartographer/cartographer/mapping/probability_values.h:20,
                 from /home/lsi2/catkin_ws2/src/cartographer/cartographer/mapping/probability_values.cc:17:
/usr/include/c++/4.8/cmath:102:11: error: ‘::acos’ has not been declared
   using ::acos;
           ^
/usr/include/c++/4.8/cmath:121:11: error: ‘::asin’ has not been declared
   using ::asin;
           ^
/usr/include/c++/4.8/cmath:140:11: error: ‘::atan’ has not been declared
   using ::atan;
           ^
/usr/include/c++/4.8/cmath:159:11: error: ‘::atan2’ has not been declared
   using ::atan2;
           ^
/usr/include/c++/4.8/cmath:180:11: error: ‘::ceil’ has not been declared
   using ::ceil;
           ^
/usr/include/c++/4.8/cmath:199:11: error: ‘::cos’ has not been declared
   using ::cos;
           ^
/usr/include/c++/4.8/cmath:218:11: error: ‘::cosh’ has not been declared
   using ::cosh;
           ^

```

SirVer, 2016-11-09:
You do not have Eigen installed. Did you install everything mentioned in https://google-cartographer.readthedocs.io/en/latest/index.html#system-requirements?

Kailegh, 2016-11-10:
yes, I installed everything, and I do not understand that error because I have the files it is looking for:

```
/usr/include$ ls eigen3/Eigen/
Array           Eigen          IterativeLinearSolvers  OrderingMethods  Sparse          SPQRSupport  SuperLUSupport
Cholesky        Eigen2Support  Jacobi                  PardisoSupport   SparseCholesky  src          SVD
CholmodSupport  Eigenvalues    LeastSquares            PaStiXSupport    SparseCore      StdDeque     UmfPackSupport
Core            Geometry       LU                      QR               SparseLU        StdList
Dense           Householder    MetisSupport            QtAlignedMalloc  SparseQR        StdVector

```

SirVer, 2016-11-10:
Please post a full [gist](https://gist.github.com/) of the full build output - starting from cloning the repositories. Before running any commands, please make sure to `export VERBOSE=1` in your shell.