cartographer-project/cartographer_ros #38: Build and install "gflags_catkin" error

carl1218, 2016-09-01:
when i bulid and install "gflags_catkin", I get the following error:

==> ninja -j1 -l1 in '/home/robot/catkin_ws/build_isolated/gflags_catkin'
ninja: error: '/home/robot/catkin_ws/devel_isolated/gflags_catkin/lib/libgflags.so', needed by '/home/robot/catkin_ws/devel_isolated/gflags_catkin/lib/libgflags_catkin.so', missing and no known rule to make it
<== Failed to process package 'gflags_catkin': 
  Command '['/home/robot/catkin_ws/install_isolated/env.sh', 'ninja', '-j1', '-l1']' returned non-zero exit status 1
Reproduce this error by running:
==> cd /home/robot/catkin_ws/build_isolated/gflags_catkin && /home/robot/catkin_ws/install_isolated/env.sh ninja -j1 -l1
Command failed, exiting.

KaijenHsiao, 2016-09-01:
Same error here.

KaijenHsiao, 2016-09-01:
Works fine without the --use-ninja flag, though.

damonkohler, 2016-09-02:
There are unused repos in the rosinstall file that do not build with Ninja. This is addressed by #42. You'll have to remove those repos manually or start a new workspace from scratch.

damonkohler, 2016-09-02:
Fixed by #42.

closed 2016-09-02 by damonkohler