cartographer-project/cartographer_ros #40: Add support for rotating, planar LIDAR

damonkohler, 2016-09-01:
The 3D mode wants things via a a PointCloud2 topic and requires externally adding a conversion from scan data to PointCloud2. To remove that dependency and for convenience, supporting plain LaserScan input for 3D could be beneficial.

Additionally, this would allow Cartographer to handle the aggregation of partial revolutions, similar to how Velodyne data is handled.

damonkohler, 2016-10-05:
Upon further consideration, this really only requires a node that converts LaserScan messages into PointCloud2 messages. Everything else should just work TM. @skohlbr, thoughts?

skohlbr, 2016-10-05:
In principle yes, there are some detail questions about the target frame used for the conversion and resulting accuracy.

For highest accuracy, one would like to use [high fidelity projection](http://wiki.ros.org/laser_geometry#High_fidelity_projection), taking motion of the LIDAR during scan acquisition into account. If one assumes odometry is good, it would make sense to aggregate scan points in the odom frame, but the sensor pose is lost then.

Another option is to convert to a cloud in a non-rotating frame relative to the spinning LIDAR to compensate for scan distortion from rotation and then provide this to cartographer. This requires the robot to provide a suitable non-rotating frame however.

The easiest option is to just use [simple projection](http://wiki.ros.org/laser_geometry#Simple_projection) and ignore scan distortion from rotation. This will likely introduce a bias in the map over time, however.

A small conversion node is this one: https://github.com/team-vigir/vigir_lidar_proc/blob/master/vigir_laserscan_to_pointcloud/src/laserscan_to_pointcloud_node.cpp
If `use_high_fidelity_projection` is false, it will convert the scan to the same frame as the original scan using simple projection. If true, it converts to a frame given by `target_frame` using high fidelity projection.

damonkohler, 2016-10-06:
For the VLP data, we treat each UDP packet of data as rigid. The laser scan to point cloud converter could subdivide the laser scan into n segments to simulate that behavior.

skohlbr, 2016-10-06:
Yes, that should indeed be a workable solution. Our VLP hasn't arrived yet, was not aware that it delivers data in such small chunks.

skohlbr, 2016-10-24:
Small heads up, created a fully working example setup of a robot with a spinning LIDAR here: https://github.com/tu-darmstadt-ros-pkg/cartographer_hector_tracker
Launch setup currently uses aforementioned [vigir_laserscan_to_pointcloud](https://github.com/team-vigir/vigir_lidar_proc/blob/master/vigir_laserscan_to_pointcloud/src/laserscan_to_pointcloud_node.cpp) to convert from scan to cloud. 

There is a simulation-based bag file example available, but I'm also preparing a real robot dataset.

athtest800, 2016-10-25:
Without odom it simply give bad results.

commit 47969496 (2016-11-22): Improves parameter names and fixes #40. (#181)
closed 2016-11-22 by damonkohler