cartographer-project/cartographer_ros #48: An error occurred while I was running 2D demo

carl1218, 2016-09-02:
when i was running 2d demo bag, I get the following erro:

process[robot_state_publisher-2]: started with pid [17777]
ERROR: cannot launch node of type [cartographer_ros/cartographer_node]: can't locate node [cartographer_node] in package [cartographer_ros]
process[rviz-4]: started with pid [17807]
process[playbag-5]: started with pid [17810]
[ERROR] [1472825523.736244580, 1432647017.705740264]: PluginlibFactory: The plugin for class 'Submaps' failed to load.  Error: Could not find library corresponding to plugin Submaps. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

damonkohler, 2016-09-02:
Sounds like you didn't build everything first?

carl1218, 2016-09-02:
I am completely in accordance with the latest version of the tutorial.

rohbotics, 2016-09-03:
You need to run the `source install_isolated/setup.bash` command in every terminal that you want to use cartographer in. This could be documented better.

carl1218, 2016-09-03:
Thank you,it was my mistake，i have run the
echo "source ~/install_isolated/setup.bash" >> ~/.bashrc
source ~/.bashrc

But,when i was running 2d demo bag, I get another erro：
[cartographer-3] process has died [pid 4946, exit code -6, cmd /home/robot/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/robot/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer __log:=/home/robot/.ros/log/002a0b9e-718c-11e6-a06c-000c29e30f9c/cartographer-3.log].
log file: /home/robot/.ros/log/002a0b9e-718c-11e6-a06c-000c29e30f9c/cartographer-3*.log

carl1218, 2016-09-03:
@rohbotics ，hi，I get another erro：

cartographer_node: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:78: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & 0xf) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
[cartographer-3] process has died [pid 4946, exit code -6, cmd /home/robot/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/robot/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer __log:=/home/robot/.ros/log/002a0b9e-718c-11e6-a06c-000c29e30f9c/cartographer-3.log].
log file: /home/robot/.ros/log/002a0b9e-718c-11e6-a06c-000c29e30f9c/cartographer-3*.log

wohe, 2016-09-08:
Sounds like an alignment issue. Are you running on a 32-bit system? I'm asking because it should not be an issue on 64-bit systems. If you're not: processor? OS? compiler?

damonkohler, 2016-10-12:
@carl1218 I'm closing this issue now since it seems clear that this is an alignment issue from running on 32-bit. This incompatibility is now documented in our system requirements.

closed 2016-10-12 by damonkohler

dadawang, 2016-10-18:
I was running 2D demo bag and got this issue:
- ERROR: cannot launch node of type [robot_state_publisher/robot_state_publisher]: robot_state_publisher
   ROS path [0]=/opt/ros/indigo/share/ros

wohe, 2016-10-18:
This doesn't seem Cartographer related: See http://answers.ros.org/questions/
That said: Did you follow our installation instructions? Is ros-indigo-robot-state-publisher installed on your system? Did you source /opt/ros/indigo/setup.bash?

herogan2017, 2019-02-22:
hello, when I run 2d-demo on my computer,found following errors. Can you me tell me why? thanks.

[ERROR] [1550813350.043405211, 1432647018.956007067]: PluginlibFactory: The plugin for class 'Submaps' failed to load.  Error: According to the loaded plugin descriptions the class Submaps with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped

innovatim, 2019-03-24:
@herogan2017  Have you solved the problem now? I just got the same problem, but I have already sourced the `setup.bash`.

TSRyan, 2019-10-09:
I had the same error message as @herogan2017, and this thread was the top hit on Google.  In my case, I had built only the `cartographer_ros` package, neglecting to build the `cartographer_rviz` package.  After building `cartographer_rviz` and sourcing the proper `setup.bash`, it worked fine.

JankinJiang, 2020-02-11:
@herogan2017 I met this question just now. It needs run command "sudo apt-get install ros-kinetic-cartographer_rviz" to use rviz in this process. and now it works fine. By the way, the cartographer package can be install directly now.