cartographer-project/cartographer_ros #86: Spinning-Rotating 2D Lidar

athtest800, 2016-10-07:
Hi!
Thank you for this project!
I would like to ask, the imu is on the rotating-spinning lidar or on base robot frame?
Thank you!

damonkohler, 2016-10-07:
I'm not sure exactly what you're asking for. There's no specific support for a rotating planar LIDAR at the moment, see #40.

If you're referring to sample data, the URDFs, which describe the sensor configuration, are available for everything. We have no sample data with a rotating planar LIDAR.

athtest800, 2016-10-07:
Thanks for the response!
I mean a rotating Hokuyo 30LX.

damonkohler, 2016-10-07:
In that case, there's no special support for that currently. You can configure it however you see fit. Refer to #40 for suggestions on how use the data with Cartographer.

closed 2016-10-07 by damonkohler