cartographer-project/cartographer_ros #87: cartographer_node crash at Optimizing 74.4%

josephduchesne, 2016-10-07:
This has happened twice now on different data sets:
```
Optimizing: 74.4%...terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[cartographer_node-4] process has died [pid 11556, exit code -6, cmd /home/joseph/Dev/google-cartographer/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/joseph/Dev/google-cartographer/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename test_2d.lua __name:=cartographer_node __log:=/home/joseph/.ros/log/54cb7c3a-8bf1-11e6-a908-d8cb8a97cc31/cartographer_node-4.log].
```

damonkohler, 2016-10-07:
It's likely that you're running out of memory. How much memory is on your system? Could you have a look at what top says about usage?

josephduchesne, 2016-10-07:
I have 16GB of RAM + 4GB of swap. I'll see what I can do, although the "Optimizing" phase takes a very long time (10+ hours), so it's hard to catch the exact moment when this occurs.

damonkohler, 2016-10-07:
10+ hours of optimizing sounds like something is wrong. Are you running over sample data? Did you somehow compile in non-optimized mode?

josephduchesne, 2016-10-07:
That's good to hear that 10+ hours is too much. I agree!

I am running on my own data, but it's fairly similar to the sample data (Hokoyo UST-20LX at 40Hz + wheel encoder odometry at 40Hz). ~40 minutes of data.

I'm building with catkin_make_isolated --install and the only configuration file changes I've made are switching from multi-echo to "use_horizontal_laser = true" and disabling imu.

damonkohler, 2016-10-07:
Could you share some more specs from your system? Can you reproduce the behavior with our demo bags (all should run in real-time)?

josephduchesne, 2016-10-07:
System specs: i7-4790K 4.4Ghz, 16GB DDR3-1600, 128GB SSD, GTX 750TI GPU, OS: Ubuntu 14.04 with Pantheon window manager.

I should add, this occurs when I call the /finish_trajectory service. 

Running 2 minutes of
`roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag`
then calling 
`rosservice call /finish_trajectory "stem: 'test'"`
results in about 9 minutes of "Optimizing: XX.X%..."

Top Results by completion percent:
Memory usage at 57.5%: 1388MB

Percent Complete | Memory Usage
------------ | -------------
57.5% | 1388MB
69.0% | 1589MB
92.0% | 2004MB
Done | 2155MB

wohe, 2016-10-07:
This is on a 64-bit system, correct? Did you change any options/code when running the demo bag? How do you measure memory usage, i.e. is this the "RES" column? 

For comparison: On my machine, stopping after the first 2 minutes of this bag takes less than 400 MB and the "Optimizing" part at the end is done after a few seconds.

josephduchesne, 2016-10-07:
This is a 64 bit system, and memory usage is the RES column from `top`. 

uname -a results:
Linux TARS 4.4.0-34-generic #53~14.04.1-Ubuntu SMP Wed Jul 27 16:56:40 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

I'm going to try running it in a docker container with a fresh 14.04 instance to rule out some sort of weird compiler or library configuration on my end.

josephduchesne, 2016-10-07:
I got google cartographer to compile and run in a fresh 14.04 docker container and it did not have this problem. It must be some sort of configuration or build issue on my local computer. 

Thank you for your help!

(And if you have a moment, is there any way to get the output after Optimizing from the ROS node? I can't seem to figure out how to get it to save an image).

closed 2016-10-07 by josephduchesne