cartographer-project/cartographer_ros #89: finish_trajectory service not saving map when called

josephduchesne, 2016-10-07:
I'm calling the service with:
`rosservice call /finish_trajectory "stem: 'test'"`

However I receive 
> ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details

On the Cartographer node:
> Optimizing: Done.     
> Optimizing: Done.     
> [cartographer_node-5] process has finished cleanly

My .lua configuration is:
`
include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry_data = true,
  use_constant_odometry_variance = true,
  constant_odometry_translational_variance = 1e-2,
  constant_odometry_rotational_variance = 1e-1,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.1,
  horizontal_laser_max_range = 30.,
  horizontal_laser_missing_echo_ray_length = 5.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true

return options
`

damonkohler, 2016-10-10:
The finish_trajectory service also shuts down the node. If that occurs before the service can reply (which should almost certainly be the case), you'll see that error. However, the map is still saved to disk. Could you verify that the map wasn't saved to disk?

wohe, 2016-10-10:
Just for clarification: If you did not specify a different `ROS_HOME`, I'd expect the map to be at `~/.ros/test.pgm` and `~/.ros/test.yaml`

josephduchesne, 2016-10-11:
I can confirm that the map is being saved to ~/.ros/test.*

This story is either a minor documentation problem, or a minor bug.

I was looking for the log output from:
`LOG(INFO) << "Saving map to '" << filename << "'...";`
but it doesn't appear anywhere on the console output or in ~/.ros, so I assumed it wasn't firing. It's probably the same race condition @damonkohler mentioned causing the service reply to error.

damonkohler, 2016-10-11:
Thanks for confirming. This will be resolved when /finish_trajectory starts a new trajectory instead of shutting down. See #98.

closed 2016-10-11 by damonkohler