cartographer-project/cartographer_ros #92: Demo failure. Not able to close loops on Muesuem data(Paper)

nitin5, 2016-10-10:
Greetings,
I am unable to run demo launch successully. Installations has been completed as per instructions.
[Console output.txt](https://github.com/googlecartographer/cartographer_ros/files/519716/Console.output.txt)
[image]
Please advice.
Rgds,

damonkohler, 2016-10-10:
Looks to me like the background optimization fell severely behind. Reminds me of #87.

Could you try calling the finish_trajectory service and see what happens? If it takes a long time to return, then I think you've somehow not compiled everything in optimized mode. Building and running in a Docker container (as in #87) should fix it.

That said, I'd like to understand what about your system has caused this to happen. Could you give us some specs?

nitin5, 2016-10-10:
I am using Ubuntu16.04 (x86_64) on Intel Xeon processor with 8GB RAM and 8GB swap space. 
I will try out listed option and share my findings.
Thank you.

chengwei0427, 2016-10-11:
Hi,@nitin5 
I get the same problem before,I think it should be not installed successfully. Then I reinstall it , and get a good result now. hope helpful!

damonkohler, 2016-10-11:
* On your 14.04 system, it seems that your version of Eigen is incorrect. The correct version of Eigen for 14.04 is 3.2.0-8. Perhaps you manually installed a different version?
* The required hardware for real-time performance is any relatively modern PC.
* Optimizations happen in the background after a fixed number of scans have been added to the map. Multiple optimizations may pile up if they're not fast enough. This should happen rarely on a properly configured system. Since optimizations only occur every n-scans, you can use the /finish_trajectory service to perform a final optimization after all bag data has been consumed.

nitin5, 2016-10-12:
Greetings,
I have completed installation after a fresh installation on Ubuntu 16.04. 

On my laptop, it still misses optimization. After call to finish_trajectory service, correct map is generated.

damonkohler, 2016-10-13:
I assume your laptop is the system running 16.04. How long does /finish_trajectory take before the map is generated? If it's only a second or so, I don't see any issues.

On the 14.04 system, it's clear your libraries do not match the system defaults. You can try building and running in a Docker container (documentation coming, see googlecartographer/cartographer:#53) if you don't want to make changes to your system.

damonkohler, 2016-10-18:
Closing this issue as stale, please reopen if you're still having issues.

closed 2016-10-18 by damonkohler

nitin5, 2016-10-18:
Hi 
I am able to manage to establish on Ubuntu16.04 as fresh install. 

I am still trying to establish on Ubuntu 14.04 . I am getting following output

> [ INFO] [1476807937.395538142]: I1018 21:55:37.000000 10710 configuration_file_resolver.cc:40] Found '/home/nn/catkinws_cartographer2/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
> [ INFO] [1476807937.473734569]: I1018 21:55:37.000000 10710 submaps.cc:176] Added submap 1
> [ WARN] [1476807938.166291116, 1432647016.571113643]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.515060600 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.167838210, 1432647016.571113643]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.541367100 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.169188727, 1432647016.571113643]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.567927400 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.188909978, 1432647016.591298103]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.594163800 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.215239634, 1432647016.621556255]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.620664000 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.241669558, 1432647016.641745641]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.647110500 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.268083028, 1432647016.672007442]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.673534700 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.294317872, 1432647016.692175222]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.699776800 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.320257834, 1432647016.722432559]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.725666600 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.346209740, 1432647016.752699465]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.751525900 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.371948825, 1432647016.772888794]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.777363600 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.397711838, 1432647016.803218137]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.803045400 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.433175424, 1432647016.833527555]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.838595000 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.459618459, 1432647016.863803382]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.865022800 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
> [ WARN] [1476807938.486370530, 1432647016.884001634]: W1018 21:55:38.000000 10710 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1432647016.891547700 but the earliest data is at time 1432647016.982251246, when looking up transform from frame [horizontal_laser_link] to frame [base_link]
[... cut ...]
> [ WARN] [1476807965.569268365, 1432647043.970149870]: W1018 21:56:05.000000 10710 ordered_multi_queue.h:165] Queue (0, echoes) exceeds maximum size.
> [ WARN] [1476807967.211948343, 1432647045.616847356]: W1018 21:56:07.000000 10710 ordered_multi_queue.h:165] Queue (0, echoes) exceeds maximum size.

I have pre-installed Ceres solver which I am using, not one downloaded from net. I have modified CMakeList file to reflect the same. The above listed error do not appear to originate because of existing Ceres pkg. 
Thanks and Rgds
Nitin

damonkohler, 2016-10-19:
This looks like a different issue. Some of your topics are not wired up correctly or you've configured Cartographer to expect some type of data that it's not receiving.

Try running roswtf and see which topics it claims are disconnected. Double check your configuration and see that odometry and IMU data are either provided or disabled.