cartographer-project/cartographer_ros #95: Localization in Existing Map

alexbaucom17, 2016-10-11:
First, off I just wanted to say thanks for this excellent work! I am part of a research group/class at the University of Pennsylvania and we are very interested in implementing this mapping software on a fleet of service robots that we are designing. However, after looking at the blog post about this project you said that localization in an existing map is not yet supported, which will actually be very important for us. I was just curious if you had any sort of rough estimate as to when that feature would be available (i.e. on the order of weeks or months?). I just need to figure out if waiting for that feature will fit into our timeline or if we should pursue other options. Thanks!

josephduchesne, 2016-10-11:
I am also very interested in an answer to this query, and interested in knowing if there are any areas that the open source community could work on to expedite the feature's development.

damonkohler, 2016-10-11:
I encourage both of you to join our mailing list (https://groups.google.com/forum/#!forum/google-cartographer) where there's currently an outline of the work that needs to be done to support this.

That said, localization in an existing map is possible in ROS already. For example, http://wiki.ros.org/amcl or http://wiki.ros.org/mrpt_localization.

As for an ETA, I'm afraid I can't help there. Top priority right now is getting everything stable and documented. We're of course happy to accept contributions!

wohe, 2016-10-11:
In addition to what Damon already said: If you cannot wait, you can try out [PerformGlobalLocalization()](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_2d/scan_matching/fast_global_localizer.h) but be aware that we do not provide any ROS integration or documentation for it yet. Also, this is not suitable for continuously localizing, but intended to find a starting location.

alexbaucom17, 2016-10-11:
Great. Thank you for the info! I wasn't sure if the maps you were producing would be compatible with other ROS packages or not but since they are I will definitely look into some of the other localization packages.

tom8382, 2016-11-14:
has relocalization feature got worked?

damonkohler, 2016-11-14:
@tom8382 We'll update this issue as we make progress here.

damonkohler, 2017-07-20:
See googlecartographer/cartographer#315.

closed 2017-07-20 by damonkohler