cartographer-project/cartographer_ros #101: Problems building and installing

oaa3wf, 2016-10-12:
I followed the instructions [here](https://google-cartographer-ros.readthedocs.io/en/latest/) but still have problems building. 

My problem is really after this line : 
catkin_make_isolated --install --use-ninja

Here's a couple of the errors I'm getting:

/usr/local/include/gmock/gmock-matchers.h:3451:7: error: provided for ‘template<class MatcherTuple> class testing::internal::UnorderedElementsAreMatcher’
 class UnorderedElementsAreMatcher { ...
...
/usr/local/include/gmock/gmock-generated-matchers.h:968:5: error: ‘tuple’ is not a member of ‘testing’
     ::testing::tuple< ...
...
...
<== Failed to process package 'cartographer':

My system is configured as follows:
64-bit intel  i7-4600U CPU
8GB RAM
Ubuntu 14.04 (Trusty)
gcc version 4.9.3
ROS Indigo

damonkohler, 2016-10-12:
Looks like your version of gcc isn't what normally comes on Trusty. Can you try with 4.8.4?

SirVer, 2016-10-12:
Please also post the full compile output into a gist and link it here.

oaa3wf, 2016-10-12:
Thanks for the responses. 
Here is the full compile output : [gist](https://gist.github.com/oaa3wf/6a4b0f8e20a99de854971a1b4c1fdef2)

I tried downgrading to 4.8.4 unsuccesfully. I managed to go to 4.8.5 but that didn't help. Do you have any tips on how to downgrade to 4.8.4?

SirVer, 2016-10-13:
It seems you do not have the proto compiler installed, see your gist: https://gist.github.com/oaa3wf/6a4b0f8e20a99de854971a1b4c1fdef2#file-gistfile1-txt-L88

Have you run through the [installation instructions](https://google-cartographer.readthedocs.io/en/latest/)?

oaa3wf, 2016-10-13:
I just tried following those instructions and still have problems. 
The output is linked [here](https://gist.github.com/oaa3wf/c25c12f0a440563c3e1121454a5f4e17).

Thanks for your help so far!

SirVer, 2016-10-14:
This is no the full output I had in mind, sorry for being imprecise: Please include the commandline and the prior commands you ran in the output. Also set the `VERBOSE` environment variable so we can see what commands are run. You can also follow step by step [what our test on Travis does](https://github.com/googlecartographer/cartographer_ros/blob/master/scripts/install.sh) to figure out where the individual steps differ.

nxtrobotics, 2016-10-16:
@oaa3wf,

I just ran a successful build and documented my steps - which are a combination of the cartographer_ros and non-ROS. Take a look at our steps and let me know if these help:

- [(https://github.com/nxtrobotics/pub_docs/blob/master/google_cartographer_install.md)]

Btw, the instructions are based on Ubuntu 14.04 LTS (AMD64) running ROS Indigo.

- Jeff

SirVer, 2016-10-17:
@nxtrobotics Could you craft pull requests with the additional changes that you required? I do not understand why you did require bullet for example. Also I wonder where the `test_tf2` directory in your workspace did come from - to my knowledge Cartographer does not create that.

oaa3wf, 2016-10-19:
I'm still not sure what's wrong. Maybe it is my system.
@SirVer [Here ](https://gist.github.com/oaa3wf/5a29c319c80f855cf57d8436e9fe5eeb)is  a link to the output with the commands. Hopefully this helps and is what you asked for, otherwise please let me know.
@nxtrobotics  Thanks for your suggestion. I tried it but it didn't work too.

Thanks for the help so far!

SirVer, 2016-10-20:
Yes, that is exactly the kind of output I was hoping for. Could you provide the output of `/usr/bin/c++  --version` and `dpkg -L google-mock`. Right now I assume that either the gcc compiler is not supported or that something messed with your gmock installation.

oaa3wf, 2016-10-20:
That's what I think too.  I'm more on the side of the gmock installation being messed up though.
[Here](https://gist.github.com/oaa3wf/b7d2d9da810f68b7ee1d81e99acb58ab) is the output you asked for.

For installing g-mock, I believe I used apt-get i.e.
sudo apt-get update
sudo apt-get install google-mock

SirVer, 2016-10-26:
gcc output seems fine, but something in your installation is messed up:

``
/usr/local/include/gmock/gmock-matchers.h:3451:7: error: provided for ‘template class testing::internal::UnorderedElementsAreMatcher’
```

uses gmock from `/usr/local/`, but the output of `dpkg -L google-mock` does not show files in `/usr/local/include/`, so you must have manually installed a different version of gmock on your system somehow and that confuses the build.

closed 2016-10-26 by SirVer