cartographer-project/cartographer_ros #106: Significance of tracking frame

nitin5, 2016-10-12:
Hi
I am getting following error on running code with custom bag file.

> [FATAL] [1476292525.887607468, 1474547763.288452865]: F1012 22:45:25.000000 10585 sensor_bridge.cc:94] Check failed: sensor_to_tracking->translation().norm() < 1e-5 The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise

What is significance of  tracking frame?
Rgds

damonkohler, 2016-10-13:
I had a look at your bag and it's clear that your IMU isn't oriented as your URDF states (i.e. z-axis aligned with the world). Your laser is slowly tilting to the side because that's the direction the IMU says is up. Also, the linear accelerations are incredibly noisy. You may not want to use the IMU on this platform.

Please see the [configuration documentation](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html). There it describes the meaning of the tracking frame and how it must relate to the IMU frame.

Increasing the linear search window of CSM to 0.2 seems to eliminate the sliding in the bag you provided.

SirVer, 2016-10-13:
​Tracking frame is the frame that the SLAM will track, i.e. localize in the
map it generates. It has to be colocated with your IMU, because it is
mathematically impossible to correctly translate linear accelerations into
a different frame. It can be rotated with respect to the IMU, but it has to
be in the same position.

Google Germany GmbH
Erika-Mann-Straße 33
80331 München

Registergericht und -nummer: Hamburg, HRB 86891
Sitz der Gesellschaft: Hamburg
Geschäftsführer: Matthew Scott Sucherman, Paul Terence Manicle

damonkohler, 2016-10-18:
Closing this issue as stale. Please reopen if you're still having issues.

closed 2016-10-18 by damonkohler