cartographer-project/cartographer_ros #119: /usr/bin/ld: cannot find -lQt5::Widgets

josch, 2016-10-14:
On a Debian unstable system, I run into the following problem when compiling cartographer_ros:

    [100%] Linking CXX executable ../devel/lib/cartographer_ros/cartographer_node
    /usr/bin/ld: cannot find -lQt5::Widgets
    collect2: error: ld returned 1 exit status
    cartographer_ros/CMakeFiles/cartographer_node.dir/build.make:353: recipe for target 'devel/lib/cartographer_ros/cartographer_node' failed
    make[2]: *** [devel/lib/cartographer_ros/cartographer_node] Error 1
    CMakeFiles/Makefile2:2202: recipe for target 'cartographer_ros/CMakeFiles/cartographer_node.dir/all' failed
    make[1]: *** [cartographer_ros/CMakeFiles/cartographer_node.dir/all] Error 2
    Makefile:138: recipe for target 'all' failed
    make: *** [all] Error 2

The linking against Qt5::Widgets seems to come from the vtk library as is evident from looking through my `CMakeFiles` directory and into the vtk cmake files installed on my system. A shared library namey `Qt5::Widgets` does probably not exist - instead, this is the qt cmake way of linking against the Qt5 widgets library. Indeed the problem seems to be known:

http://www.vtk.org/pipermail/vtk-developers/2014-May/030456.html

As proposed, after adding `find_package(Qt5Widgets REQUIRED)` to the `CMakeLists.txt` of cartographer_ros, the linking process finishes successfully.

It is not obvious where the bug here is, whether it's a bug in cartographer_ros, which should add a `find_package(Qt5Widgets REQUIRED)` or whether vtk should do that instead. The fact is, that currently, because of this problem, cartographer_ros does not link successfully on Debian unstable.

There also seem to be a fix for this problem in vtk:

 http://review.source.kitware.com/15372

SirVer, 2016-10-19:
@josch Could you try out https://github.com/SirVer/cartographer_ros/tree/find_qt_widgets and report if that fixes the issue?

SirVer, 2016-10-19:
And if it does not, could you try with and without the line and also with removing `set(QT_LIBRARIES Qt5::Widgets)` around the same area changed in https://github.com/SirVer/cartographer_ros/commit/470b06774bb5211f03c7d3dcd93efbcffca18e38

josch, 2016-10-19:
@SirVer Thank you, your commit fixes this problem.

SirVer, 2016-10-19:
thanks for checking.

PR #133 (2016-10-19): Manually require Qt5Widgets library if we use Qt5.
Fixes #119.
commit 7f7933f2 (2016-10-19): Manually require Qt5Widgets library if we use Qt5. (#133)

Might fix #119.
closed 2016-10-19 by SirVer