cartographer-project/cartographer_ros #120: ninja: error: /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8' needed by

look-recognize, 2016-10-15:
When I built the source by using "catkin_make_isolated --install --use-ninja", there is an error:
****************************************Error info****************************************
ninja: error: '/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8', needed by '/home/cartographer/devel_isolated/cartographer_rviz/lib/libcartographer_rviz_submaps_visualization.so', missing and no known rule to make it.
*********************************************************************************************

Is opencv a necessary dependency for rviz?
How can I solve it?

SirVer, 2016-10-24:
OpenCV is not a dependency. Please provide the full build output starting from cloning the repositories and set `VERBOSE=1` in your environment.

wlongxiang, 2016-10-25:
I also got the same error on my Jetson Tk1 board

SirVer, 2016-10-25:
@wlongxiang  Please provide the full build output starting from cloning the repositories and set VERBOSE=1 in your environment. 

We do not require OpenCV, I do not know which dependency pulls it in on your sytems, so I need output or a repro case.

wlongxiang, 2016-10-25:
@SirVer I followed this installation process directly. https://google-cartographer-ros-for-turtlebots.readthedocs.io/en/latest/.
I am kind of new to ROS, could you tell me how to set VERBOSE to 1 and make a log file for the installation error? I am also googling around..

SirVer, 2016-10-25:
@wlongxiang On which Linux distribution and version are you? You can figure that out by running `lsb_release -a`.

The easiest to get full output is to do type the following in your terminal:

    script output.log
    export VERBOSE=1
    
now run the installation and once done type `exit`. The full output of everything you typed and all command output is now in `output.log`.

wlongxiang, 2016-10-25:
@SirVer sorry for the character coding problem in the log file, 
outout for `lsb_release -a` is:
`No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.4 LTS
Release:	14.04
Codename:	trusty
`
and this is the output.log:
`Script started on Tue Oct 25 16:13:49 2016
]0;ubuntu@tegra-ubuntu: ~/catkin_wsubuntu@tegra-ubuntu:~/catkin_ws$ export VERBOSE=1
]0;ubuntu@tegra-ubuntu: ~/catkin_wsubuntu@tegra-ubuntu:~/catkin_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/ubuntu/catkin_ws
Source space: /home/ubuntu/catkin_ws/src
Build space: /home/ubuntu/catkin_ws/build_isolated
Devel space: /home/ubuntu/catkin_ws/devel_isolated
Install space: /home/ubuntu/catkin_ws/install_isolated
[35m~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[0m
[35m~~[0m  traversing 6 packages in topological order:[0m
[35m~~[0m  - [1m[34mceres_solver[0m ([1m[36mplain cmake[0m)[0m
[35m~~[0m  - [1m[34mcartographer[0m ([1m[36mplain cmake[0m)[0m
[35m~~[0m  - [1m[34mcartographer_ros_msgs[0m[0m
[35m~~[0m  - [1m[34mcartographer_ros[0m[0m
[35m~~[0m  - [1m[34mcartographer_rviz[0m[0m
[35m~~[0m  - [1m[34mcartographer_turtlebot[0m[0m
[35m~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~[0m

]2;ceres_solver [1 of 6][1m[32m==>[0m [0mProcessing [36mplain cmake[0m package: '[1m[34mceres_solver[0m'[0m
build.ninja exists, skipping explicit cmake invocation...
[1m[34m==>[0m[1m ninja build.ninja[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/ceres_solver/install[0m'[0m
ninja: no work to do.
[0m[1m[34m==>[0m[1m ninja -j1 -l1[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/ceres_solver/install[0m'[0m

[1/5] Performing update step for 'ceres_src'[K
[1/5] Performing update step for 'ceres_src'[K
[1m[34m==>[0m[1m ninja install[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/ceres_solver/install[0m'[0m

[1/6] Performing update step for 'ceres_src'[K
[1/6] Performing update step for 'ceres_src'[K
[2/2] Install the project...[K
[2/2] Install the project...[K
-- Install configuration: ""
[32m<==[0m Finished processing package [[32m[1m1[0m of [1m[32m6[0m]: '[34m[1mceres_solver[0m'[0m

]2;cartographer [2 of 6][1m[32m==>[0m [0mProcessing [36mplain cmake[0m package: '[1m[34mcartographer[0m'[0m
[1m[34m==>[0m[1m Building with env: '/home/ubuntu/catkin_ws/install_isolated/env.sh'[0m
build.ninja exists, skipping explicit cmake invocation...
[1m[34m==>[0m[1m ninja build.ninja[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/cartographer/install[0m'[0m
ninja: no work to do.
[0m[1m[34m==>[0m[1m ninja -j1 -l1[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/cartographer/install[0m'[0m

[1/1] Building documentation.[K
[1/1] Building documentation.[K
Running Sphinx v1.2.2
loading pickled environment... done
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.
[1m[34m==>[0m[1m ninja install[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/cartographer/install[0m'[0m

[1/2] Building documentation.[K
[1/2] Building documentation.[K
Running Sphinx v1.2.2
loading pickled environment... done
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 0 changed, 0 removed
looking for now-outdated files... none found
no targets are out of date.

[... 296 lines cut ...]
-- Up-to-date: /home/ubuntu/catkin_ws/install_isolated/include/cartographer_ros/sensor_bridge.h
-- Up-to-date: /home/ubuntu/catkin_ws/install_isolated/include/cartographer_ros/tf_bridge.h
-- Up-to-date: /home/ubuntu/catkin_ws/install_isolated/include/cartographer_ros/time_conversion.h
-- Up-to-date: /home/ubuntu/catkin_ws/install_isolated/bin/cartographer_node
-- Up-to-date: /home/ubuntu/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_node
[32m<==[0m Finished processing package [[32m[1m4[0m of [1m[32m6[0m]: '[34m[1mcartographer_ros[0m'[0m

]2;cartographer_rviz [5 of 6][1m[32m==>[0m [0mProcessing [36mcatkin[0m package: '[1m[34mcartographer_rviz[0m'[0m
[1m[34m==>[0m[1m Building with env: '/home/ubuntu/catkin_ws/install_isolated/env.sh'[0m
build.ninja exists, skipping explicit cmake invocation...
[1m[34m==>[0m[1m ninja build.ninja[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/cartographer_rviz[0m'[0m
ninja: no work to do.
[0m[1m[34m==>[0m[1m ninja -j1 -l1[0m [1m[30min[0m '[1m/home/ubuntu/catkin_ws/build_isolated/cartographer_rviz[0m'[0m
ninja: error: '/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by '/home/ubuntu/catkin_ws/devel_isolated/cartographer_rviz/lib/libcartographer_rviz_submaps_visualization.so', missing and no known rule to make it
[31m[1m<==[0m Failed to process package '[1m[34mcartographer_rviz[0m': 
  Command '['/home/ubuntu/catkin_ws/install_isolated/env.sh', 'ninja', '-j1', '-l1']' returned non-zero exit status 1[0m

[31mReproduce this error by running:[0m
[32m[1m==> [0m[0mcd /home/ubuntu/catkin_ws/build_isolated/cartographer_rviz && /home/ubuntu/catkin_ws/install_isolated/env.sh ninja -j1 -l1

Command failed, exiting.
]0;ubuntu@tegra-ubuntu: ~/catkin_wsubuntu@tegra-ubuntu:~/catkin_ws$ [Kexit
exit

Script done on Tue Oct 25 16:14:33 2016`

SirVer, 2016-10-26:
I ran a pristine trusty docker image for the following analysis. I ran through [the ROS installation instructions](https://google-cartographer-ros.readthedocs.io/en/latest/?badge=latest#building-installation) and investigated where the file comes from:

This file is pulled in by the package `libopencv-videostab2.4`:
~~~
$ apt-file search /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8
libopencv-videostab2.4: /usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8
~~~

Trying to remove this reveals that indeed, `ros-indigo-rviz` depends on this package:

~~~
$ sudo apt-get remove libopencv-videostab2.4
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libopencv-dev libopencv-videostab-dev libopencv-videostab2.4
  ros-indigo-image-geometry ros-indigo-rviz
0 upgraded, 0 newly installed, 5 to remove and 14 not upgraded.
After this operation, 11.1 MB disk space will be freed.
Do you want to continue? [Y/n] 
~~~

After installing `ros-indigo-robot-state-publisher` and [X11 forwarding into my docker](http://fabiorehm.com/blog/2014/09/11/running-gui-apps-with-docker/) I was able to run the 2D demo as described in the documentation.

I cannot say why this file is not installed on your systems - it is installed in the `rosdep install` phase on my pristine image. 

Closing, as this is not Cartographer related but a ROS installation problem. Please reopen if I missed something.

closed 2016-10-26 by SirVer