cartographer-project/cartographer_ros #121: I want to use cartographer_ros with one hokuyo laser

carl1218, 2016-10-16:
hi,
  Thank you for this project!
   I only use one HOKUYO laser, and no other robot hardware.Try to record a file bag.then runed this
file bag for mapping,but I get some error.
   I runed the hokuyo.launch to record the data.then i runed the demo,but nothing in the rviz,and in the rviz prompt error ：Globl Status was warn (Fixed Frame:No tf data.  Actual error: Fixed Frame [map] does not exist)。
   I would like to ask，the cartographer_ros must cooperate with the robot drive？look like turtlebot .
 Can i only used the 2D Lidar for maping?

I modified the original launch configuration ：
**The lua file is here:**
 include "map_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.,
    horizontal_laser_max_range = 30.,
    horizontal_laser_missing_echo_ray_length = 5.,
    constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)
return options

 **The hokuyo launch is here**:
<launch>
  <node name="hokuyo_node" pkg="hokuyo_node" type="hokuyo_node">
    <param name="port" type="string" value="/dev/ttyACM0"/>
    <param name="min_ang" value="-1.7" />
    <param name="max_ang" value="1.7" />
    <param name="min_range" value="0.18" />
    <param name="hokuyo_node/calibrate_time" value="true" />
    <param name="tracking_frame" type="string" value="laser_link" />
    <remap from="scan" to="/laser_scan"/>
  </node>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />

**The demo launch is here**
<launch> 
<param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>

carl1218, 2016-10-16:
I am so sorry, lose some of the information from the hokuyo launch and demo launch
The hokuyo launch is here:
<launch>
  <node name="hokuyo_node" pkg="hokuyo_node" type="hokuyo_node">
    <param name="port" type="string" value="/dev/ttyACM0"/>
    <param name="min_ang" value="-1.7" />
    <param name="max_ang" value="1.7" />
    <param name="min_range" value="0.18" />
    <param name="hokuyo_node/calibrate_time" value="true" />
    <param name="tracking_frame" type="string" value="laser_link" />
    <remap from="scan" to="/laser_scan"/>
  </node>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />
  </node>
</launch>

The demo launch is here:
<launch> 
<param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>

carl1218, 2016-10-16:
I  am so sorry , Partial loss of information from the hokuyo launch and demo launch
**The hokuyo launch is here:**
`<launch>
  <node name="hokuyo_node" pkg="hokuyo_node" type="hokuyo_node">
    <param name="port" type="string" value="/dev/ttyACM0"/>
    <param name="min_ang" value="-1.7" />
    <param name="max_ang" value="1.7" />
    <param name="min_range" value="0.18" />
    <param name="hokuyo_node/calibrate_time" value="true" />
    <param name="tracking_frame" type="string" value="laser_link" />
    <remap from="scan" to="/laser_scan"/>
  </node>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />
  </node>
</launch>`

**The demo launch is here:**
`<launch> 
<param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>
  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>`

carl1218, 2016-10-16:
I am so sorry , Partial loss of information from the hokuyo launch and demo launch
**The hokuyo launch is here:**
```
<launch>
  <node name="hokuyo_node" pkg="hokuyo_node" type="hokuyo_node">
    <param name="port" type="string" value="/dev/ttyACM0"/>
    <param name="min_ang" value="-1.7" />
    <param name="max_ang" value="1.7" />
    <param name="min_range" value="0.18" />
    <param name="hokuyo_node/calibrate_time" value="true" />
    <param name="tracking_frame" type="string" value="laser_link" />
    <remap from="scan" to="/laser_scan"/>
  </node>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />
  </node>
</launch>
```
**The demo launch is here：**
```
<launch> 
<param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>
```

damonkohler, 2016-10-17:
You can certainly map without any additional hardware. Keep in mind that without an IMU you'll need to keep your LIDAR as level with the ground plane as possible.

It looks like you have an issue with our frames. You need to set tracking_frame and published_frame to "laser_link".

carl1218, 2016-10-17:
@damonkohler 
Thank you very much.

damonkohler, 2016-10-17:
I assume that resolved your issue? If not, please reopen.

closed 2016-10-17 by damonkohler

carl1218, 2016-10-17:
@damonkohler 
Yes, this problem has been solved. thank you very much.

carl1218, 2016-10-19:
@damonkohler 
   hi, I can't make the good result using my rosbag.I was doing the test in a rectangular room(No mirrors). I did not use the IMU.But the map is not perfect.There are a lot of redundant laser beams around the map.Is it because I did not use the IMU ?
The picture is here:

[image]
[image]

damonkohler, 2016-10-19:
This looks like a different issue. From the screen shots, it seems you have the [missing_echo_ray_length](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) set too large for your sensor. It's not getting returns and inserting empty space. Try setting it to 0.5 or so.

josephduchesne, 2016-10-19:
@carl1218 Which Lidar model are you using? That may help people suggest better parameters for you.

carl1218, 2016-10-19:
@josephduchesne 
I used hokuyo urg 04lx

SirVer, 2016-10-19:
If you want more help in debugging you have to post your full configuration and a bag with data somewhere so others can reproduce.

carl1218, 2016-10-19:
@damonkohler 
Thank you ery much.this problem has been solved.