cartographer-project/cartographer_ros #126: How to provided tf transforms from map to odom ? just as gmapping did to correct the accumulated error of the odometer

simplePig, 2016-10-19:
Owing to the accumulated errors of the odometer calculation, the map construction is misplaced and error, but the gmapping package provides an algorithm for reversely correcting odometer data by providing a continuous transition from map to odom, the map construction is very accurate
Whether or not a similar implementation is provided in the cartographer's package, it would be impossible to build an accurate map using odometer without such a conversion transform

damonkohler, 2016-10-19:
I believe you're looking for the ["provide_odom_frame" option in the configuration](http://google-cartographer-ros.readthedocs.io/en/latest/configuration.html). Assuming you don't have some other node providing the odom frame, you can enable that option to ask Cartographer to provide a continuous odom frame.

Providing odometry to Cartographer is, however, orthoganal to asking Cartographer to provide the odom frame. See the "use_odometry_data" option.

Please reopen if you have any issues.

closed 2016-10-19 by damonkohler

simplePig, 2016-10-19:
Sorry, probably I did not describe the problem clearly.
Now, i have a small differential driving car with a scanning radius of 6 meters of laser radar. I want to draw a map of an area of 150 square meters. However, i got some mapping error due to radar too short scanning range radius using cartographer. So I would like to use odometry to make mapping more accurate less error. I have successfully provided the odometer data to the nav_msgs / Odometry node, then set use_odometry_data = true, and then ran the cartographer, but i found that there were more errors. 
I do not know which of the cartographer's algorithms is the primary reference: the lidar or the odometer, or both? After the introduction of odometer data, SLAM algorithm is local matching or global matching?