cartographer-project/cartographer_ros #135: Difference in output using Odom

1473966355, 2016-10-20:
Hi,

Now, i have a small differential driving car with a scanning radius of 6 meters of laser radar. I want to draw a map of an area of 150 square meters. However, i got some mapping error due to radar too short scanning range radius using cartographer. So I would like to use odometry to make mapping more accurate less error. I have successfully provided the odometer data to the nav_msgs / Odometry node, then set use_odometry_data = true, and then ran the cartographer, but i found that there were more errors. 

when i don't use odometry_data and use the config file https://drive.google.com/open?id=0B2-B3l8213hNejFDSVRKS0gtZ0U, the data record is the bag https://drive.google.com/open?id=0B2-B3l8213hNODk4ZWlwazFjUjA,and launch files https://drive.google.com/open?id=0B2-B3l8213hNVVJuOEtYNWltRVk .The final effect is relatively good, as follow,
[image]

but, when i use odometry_data and use the config file https://drive.google.com/open?id=0B2-B3l8213hNX3FZLXhmX2JDcms, the data record is the bag https://drive.google.com/open?id=0B2-B3l8213hNVVJuOEtYNWltRVk, and launch files https://drive.google.com/open?id=0B2-B3l8213hNVVJuOEtYNWltRVk. The final effect is bad, as follow,
[image]

I do not know which of the cartographer's algorithms is the primary reference: the lidar or the odometer, or both? After the introduction of odometer data, SLAM algorithm is local matching or global matching?

damonkohler, 2016-10-20:
There's no primary reference, per say. All of the data is fused together with different weights. That said, scan matching is usually the most important.

Including odometry doesn't change the scan matching behavior. Loop closure is still done and it is still done with scan matching.

It looks like your odometry fundamentally disagrees with scan matching.  Check that your URDF is correct and that your odometry data is being computed/published correctly.

It's also possible that your odometry data just isn't very good in translation, rotation, or both. You can try reducing the importance of it by tweaking the [sensor_bridge.odometry_xxx_variance](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) configuration options (make them larger).

rohbotics, 2016-10-20:
It looks like you might have a timing offset between your odom and your lidar.
What lidar are you using? Most have an option to sync up the the timestamps with system time.

rohbotics, 2016-10-21:
Just checked your bag file, your Odom covariance is 0.0 for all the elements. 
This will break the sensor fusion, basically making rely entirely on odom. I would set it to something between 0.1-0.3 to start with.

damonkohler, 2016-10-24:
The covariance specified in the Odometry messages are unused. Covariances should be specified using the parameters described in the [ROS API](http://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html).

That said, it is difficult to tune the system to handle the situation where scan matching fails. This is something we're working to improve by replacing the UKF with a separate optimization.

simplePig, 2016-10-25:
@damonkohler 
I have found two observation and one prediction in **pose_traker**, the computing of 2 observations according to weights how to work and where is the **Kraft's paper** as introduced ?
I think the **delta_translation** and the **delta_orientation** should be  computed differently not the same one set of algorithm.

damonkohler, 2016-10-25:
@simplePig I'm not sure I follow your argument. However, perhaps it's moot since we're working to replace the UKF with an optimization.

I'm closing this now as stale. Please reopen if there's still an issue.

closed 2016-10-25 by damonkohler