cartographer-project/cartographer_ros #138: Error building cartographer_rviz after latest update

masterhui, 2016-10-21:
Hi
I updated cartographer_ros from a previous version (Oct 6, commit 1418902493115cad90d6ee93877c5a602a78614f) to the most up-to-date one (Oct 21, commit 20487028ed36c558336a873860cfe90340e8662f). While building completed without errors previously, I now get the following error:

> Errors     << cartographer_rviz:make /home/ros/catkin_ws/logs/cartographer_rviz/build.make.005.log                                      
In file included from /home/ros/catkin_ws/build/cartographer_rviz/cartographer_rviz/../../../src/overlay/cartographer_ros/cartographer_rv
iz/cartographer_rviz/drawable_submap.h:35:0,
                 from /home/ros/catkin_ws/build/cartographer_rviz/cartographer_rviz/moc_drawable_submap.cpp:9,
                 from /home/ros/catkin_ws/build/cartographer_rviz/cartographer_rviz/drawable_submap_automoc.cpp:2:
/opt/ros/kinetic/include/rviz/display_context.h:34:19: fatal error: QObject: No such file or directory
compilation terminated.
make[2]: *** [cartographer_rviz/CMakeFiles/drawable_submap.dir/drawable_submap_automoc.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ros/catkin_ws/src/overlay/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.h:35:0,
                 from /home/ros/catkin_ws/src/overlay/cartographer_ros/cartographer_rviz/cartographer_rviz/drawable_submap.cc:17:
/opt/ros/kinetic/include/rviz/display_context.h:34:19: fatal error: QObject: No such file or directory
compilation terminated.
make[2]: *** [cartographer_rviz/CMakeFiles/drawable_submap.dir/drawable_submap.cc.o] Error 1
make[1]: *** [cartographer_rviz/CMakeFiles/drawable_submap.dir/all] Error 2
make: *** [all] Error 2

My setup:
Ubuntu 16.04
ROS Kinetic
x86_64

I do realize that you mention ROS Kinetic not being fully supported by cartographer_ros, however as there were no build errors previously and now there are, I thought I'd report this anyway.

Regards

SirVer, 2016-10-21:
Could you bisect which was the breaking build in `cartographer_ros`?

masterhui, 2016-10-21:
No, sorry, I updated straight from 1418902493115cad90d6ee93877c5a602a78614f to 20487028ed36c558336a873860cfe90340e8662f.

wuhaixing, 2016-10-23:
I meet the error too:

> /opt/ros/kinetic/include/rviz/display_context.h:34:19: fatal error: QObject:No such file or directory
compilation terminated.

SirVer, 2016-10-24:
@wuhaixing Can you point us to the breaking commit?

spyke555, 2016-10-25:
I can get it to build fine up until the Oct12th updates. Using cartographer commit 8f64860 and cartographer_ros 0e8d219 works fine. Using cartographer_ros be7ce01 breaks the build for me with the same error as described in the OP...

jsgaobiao, 2016-10-25:
I meet the error too. :(

spyke555, 2016-10-25:
The QT version is also what I am suspecting based on what I've been able to research. I have installed QT4 on my system, but likely need to instruct the build scripts to use that instead of QT5...

SirVer, 2016-10-26:
I was able to reproduce this with a fresh docker image and was able to fix. Please verify and reopen if it is still a problem.

closed 2016-10-26 by SirVer

spyke555, 2016-10-26:
Confirming problem resolved, thanks!

masterhui, 2016-10-27:
I can also confirm that this has been fixed. Well done, cheers!

PR #150 (2016-10-26): Fix build with Qt5.
- Include Qt5 include directories into search path. Fixes #138.
- Supersedes and fixes #124.
commit de53d1f7 (2016-10-26): Fix build with Qt5. (#150)

- Include Qt5 include directories into search path. Fixes #138.
- Supersedes and fixes #124.