cartographer-project/cartographer_ros #144: Queue (0, scan) exceeds maximum size

jacobperron, 2016-10-25:
I downloaded a bag from [this dataset](http://robotics.upo.es/datasets/frog/upo/#download-wrapper), specifically [the "Sensors bag" here](http://robotics.upo.es/datasets/frog/upo/dataset_April_29_2014_16.41h.html). It contains scans from a front facing laser scanner parallel to the floor and odometry. I remapped the topics accordingly and load the bag with the following launch file:

```xml
<launch>
  <arg name="bagfile" />
  <param name="/use_sim_time" value="true" />

  <node name="cartographer_node" pkg="cartographer_ros" type="cartographer_node"
        args="
          -configuration_directory $(find cartographer_navigation)/config
          -configuration_basename cartographer_config.lua"
      output="screen">
    <remap from="scan" to="scanfront" />
    <remap from="odom" to="pose" />
  </node>

  <node name="playbag" pkg="rosbag" type="play"
        args="--clock $(arg bagfile)" />
</launch>
```

Here is a copy of my configuration file:

```lua
include "map_builder.lua"

options = { 
  map_builder = MAP_BUILDER,
  sensor_bridge = { 
    horizontal_laser_min_range = 0., 
    horizontal_laser_max_range = 30.,
    horizontal_laser_missing_echo_ray_length = 5., 
    constant_odometry_translational_variance = 0., 
    constant_odometry_rotational_variance = 0., 
  },  
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "odom",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry_data = true,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.1,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

return options
```

*cartographer_node* is subscribed to the correct topics, but I receive the following warnings (the last being repeated indefinetly) and no map is created. It doesn't seem to be related to the odometry topic since I can set `use_odometry_data` to false and get the same result. The TF tree also looks fine (sans map frame). I haven't had time to dig further.

```bash
PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    playbag (rosbag/play)

auto-starting new master
process[master]: started with pid [11572]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8c6ea576-9a5c-11e6-b5ab-acd1b8c150f9
process[rosout-1]: started with pid [11585]
started core service [/rosout]
process[cartographer_node-2]: started with pid [11593]
process[playbag-3]: started with pid [11597]
[ INFO] [1477363297.169694860]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/src/oberon_navigation/config/cartographer_config.lua' for 'cartographer_config.lua'.
[ INFO] [1477363297.170119139]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1477363297.170175552]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1477363297.170280031]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1477363297.170330291]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1477363297.170540981]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1477363297.170594428]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1477363297.170800371]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1477363297.170850246]: I1024 19:41:37.000000 11593 configuration_file_resolver.cc:40] Found '../catkin_ws/devel/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1477363297.224451375]: I1024 19:41:37.000000 11593 submaps.cc:176] Added submap 1
[ WARN] [1477363297.614288202, 1398782509.836662936]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.745911000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.614618512, 1398782509.836662936]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.770786000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.614887091, 1398782509.836662936]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.795817000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.620661721, 1398782509.836662936]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.820668000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.646507840, 1398782509.866933562]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.845570000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.671598891, 1398782509.887142529]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.870480000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.696390327, 1398782509.917399640]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation at time 1398782509.895437000, but only time 1398782509.928337702 is in the buffer, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363297.721191995, 1398782509.937499099]: W1024 19:41:37.000000 11593 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1398782509.920326000 but the earliest data is at time 1398782509.928337702, when looking up transform from frame [laserfront] to frame [base_link]
[ WARN] [1477363310.247964591, 1398782522.467980409]: W1024 19:41:50.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
[ WARN] [1477363311.456086525, 1398782523.670806252]: W1024 19:41:51.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
[ WARN] [1477363312.641146897, 1398782524.863538378]: W1024 19:41:52.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
[ WARN] [1477363313.841607745, 1398782526.057108239]: W1024 19:41:53.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
[ WARN] [1477363315.042431701, 1398782527.259637737]: W1024 19:41:55.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
[ WARN] [1477363316.233640223, 1398782528.452874894]: W1024 19:41:56.000000 11593 ordered_multi_queue.h:167] Queue (0, scan) exceeds maximum size.
```

rohbotics, 2016-10-25:
You might be waiting on getting imu data, which is not in your dataset.

I think you need to add `TRAJECTORY_BUILDER_2D.use_imu_data = false` to your configuration file.

jacobperron, 2016-10-25:
@rohbotics Thanks! Missed that parameter. Works now :+1:

closed 2016-10-25 by jacobperron

Needrom, 2016-11-13:
why I still have imu_link on tf tree even I set `TRAJECTORY_BUILDER_2D.use_imu_data =  false`

damonkohler, 2016-11-14:
@Needrom Setting the parameter only enables/disables the use of IMU data. It has no affect on your TF tree.

cyy12345, 2017-04-09:
what should I do if I want to use imu data ?