cartographer-project/cartographer_ros #154: Kinect with Google Cartographer

jckchow, 2016-10-27:
Hi,

I am struggling with configuring the code to work with two Kinect-like cameras instead of the Velodynes for 3D mapping. I also have an IMU providing the relevant data. The motion I get right now is not very smooth and it seems to be jumping around all the time. Can you spot some errors in my *.lua file?

include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.,
    horizontal_laser_max_range = 5.,
    horizontal_laser_missing_echo_ray_length = 5.,
    constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_horizontal_laser = false,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 2,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

TRAJECTORY_BUILDER_3D.scans_per_accumulation = 1

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
MAP_BUILDER.sparse_pose_graph.optimization_problem.huber_scale = 5e2
MAP_BUILDER.sparse_pose_graph.optimize_every_n_scans = 2
MAP_BUILDER.sparse_pose_graph.constraint_builder.sampling_ratio = 0.03
MAP_BUILDER.sparse_pose_graph.optimization_problem.ceres_solver_options.max_num_iterations = 10
-- Reuse the coarser 3D voxel filter to speed up the computation of loop closure
-- constraints.
MAP_BUILDER.sparse_pose_graph.constraint_builder.adaptive_voxel_filter = TRAJECTORY_BUILDER_3D.high_resolution_adaptive_voxel_filter
MAP_BUILDER.sparse_pose_graph.constraint_builder.min_score = 0.62
MAP_BUILDER.sparse_pose_graph.constraint_builder.log_matches = true

return options

damonkohler, 2016-10-28:
My first guess is that the data rate is too high. Take a look at:

https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/launch/turtlebot_depth_camera_3d.launch

You'll likely want a similar configuration.

If that's not helpful, please share a bag and point us to a repo we can clone to reproduce your results.

jckchow, 2016-10-28:
Thank you for your quick response. I still seem to observe a bias in roll. I wonder if it is caused by my IMU. I'm feeding in IMU acceleration and angular rate only based on the 3D demo bag file. I assumed the UKF will take care of estimating acc biases and performs inclination corrections by assuming some dynamical model based on the pedestrian or purely on the LiDAR feedback. Do I need to have some static periods in the beginning to let my IMU biases be observed through ZUPT? I have replaced the Kinect point cloud with motion compensated and lever-arm/boresight corrected VLP-16 data to make it more consistent with the Cartographer's input and I'm still have the same issue.

damonkohler, 2016-10-31:
We don't use ZUPTs or do any bias estimation. Instead we leave that to our IMU. If you believe your IMU has biases, you'll need to handle that on your own before feeding it into Cartographer.

A bias in roll indicates there's some bias in your accelerations or that your URDF is incorrect. The IMU tracker assumes that over time acceleration points overwhelmingly in the direction of gravity.

What sort of IMU are you using? All our 3D data is collected using a Microstrain 3DM-GX4-25.

skohlbr, 2016-10-31:
Making a IMU handle biases on it's own is non-trivial as to distinguish bias from actual angular rate, another independent source of information is required. The pose estimates from cartographer provide such a independent set of measurements that would allow for bias estimation within cartographer.

Providing the functionality within cartographer could improve performance especially when using cheaper IMUs, but I agree it's probably too complex and cluttering to add. Other option would be to write out cartographer's pose estimates and have a external filter use them as input to estimate IMU biases. It looks like this capability might be added to robot_localization in the future: https://github.com/cra-ros-pkg/robot_localization/issues/249

jckchow, 2016-10-31:
Looking at the specs for Microstrain 3DM-GX4-25 they seem to only estimate the gyroscope biases. The accelerometer biases were assumed to be small enough to not cause a signification tilt. We are testing with a Sensonor IMU. I ran my own filter to remove the acc and gyr biases as suggested, but still I'm getting inconsistent results. If I run both LiDAR systems my solution jumps back and forth between two solutions (or so it seems). If I run 3D SLAM with only the horizontal LiDAR and IMU my solution doesn't jump but looks wrong. Does the base_link and imu_link have to be different? I'm currently setting them to be the same.

wohe, 2016-11-11:
Just had a look at the configuration you sent. If you use `num_lasers_3d = 2` to get input from 2 cameras, you don't want to scan match against them separately, but after both have been combined into a single point cloud. This is controlled by `TRAJECTORY_BUILDER_3D.scans_per_accumulation` which should be at least 2 in this case. Depending on the input frequency you can also try larger values.

Not important for local SLAM quality, but since I noticed it:
`MAP_BUILDER.sparse_pose_graph.optimize_every_n_scans = 2`
is too low. This is how often the poses will be optimized globally, i.e. loop closure.

jckchow, 2016-11-11:
Thanks for the suggestion. With your recommended settings, I can now see the horizontal and vertical cloud being processed together. However, I'm still experiencing some heading jitter, which results in a poor looking map. I suspect the culprit is the IMU because if I use only the horizontal point cloud in 2D mode I get a nice map. Since we do not give the IMU covariances and do not estimate the biases, I am wondering if there is a weighting issue in the UKF?

damonkohler, 2016-11-14:
The Sensonor IMU looks to be on the high end for MEMS. The bias may not be a problem. You can of course try to tune the importance of the IMU, but that's difficult to get right and is probably better left alone. That will be easier to handle once we've removed the UKF. @skohlbr, at that point we may also be able to add bias estimation without too much added complication.

This could be a calibration issue in the alignment of your cameras or in the timing of the data coming from them. I suggest trying to use just a single camera first.

jckchow, 2016-11-16:
I have a link to my bag files posted above: https://drive.google.com/file/d/0B_OfLz3Naku8dFlPeE9GZ3Y0djA/view?usp=sharing

I just ran roslaunch cartographer_ros demo_test_backpack_3d.launch bag_filename:=test_data.bag

with the launch file from here: https://github.com/jckchow/cartographer_ros

damonkohler, 2016-11-17:
Sorry, I didn't see the bag you linked earlier.

I investigated a bit and it looks to me like the alignment of your IMU is off. I suggest checking the mount and trying to calibrate that parameter. I manually tuned your URDF a bit and improved the results dramatically, but still far from what I'd expect given the data.

Also, if you're going to use the VLP-16, you should allow Cartographer to accumulate scans for you. Convert each UDP packet into a point cloud and increase the number of scans per accumulation.

jckchow, 2016-11-17:
I did change the URDF a bit to give the LIDARs and IMU a 90 deg heading difference. As you said, the results are still worse than expected. It shouldn't be a problem that I do the motion compensation myself outside of Google Cartographer and then feeding in data frame by frame right?

damonkohler, 2016-11-18:
I meant to convey that I believe your IMU is not well aligned with your expected alignment. If I change the alignment by a few degrees in roll and pitch, I can improve the results. Put another way, your acceleration vector is not pointing straight up from what looks to be the floor in your data. You need to do some calibration.

Quality will likely also improve if you allow Cartographer to handle the motion compensation for you.

damonkohler, 2016-11-30:
I'll close this now. Please reopen it if you have any more questions.

closed 2016-11-30 by damonkohler