cartographer-project/cartographer_ros #155: error while installing cartographer on Kobuki platform

ywj447, 2016-10-28:
error was like:

Building CXX object cartograp...nary.dir/lua_parameter_dictionary.cc.o
FAILED: /usr/bin/c++    -O3 -DNDEBUG -isystem /usr/local/include -isystem /usr/include/eigen3 -I. -I.. -I../cartographer/common/{Boost_INCLUDE_DIRS}     -O3 -DNDEBUG -std=c++11 -Wall -Wpedantic -Werror=format-security -Werror=return-type -Werror=uninitialized -MMD -MT cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o -MF "cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o.d" -o cartographer/common/CMakeFiles/common_lua_parameter_dictionary.dir/lua_parameter_dictionary.cc.o -c ../cartographer/common/lua_parameter_dictionary.cc
../cartographer/common/lua_parameter_dictionary.cc: In function 'void cartographer::common::{anonymous}::QuoteStringOnStack(lua_State*)':
../cartographer/common/lua_parameter_dictionary.cc:49:24: error: 'lua_pushglobaltable' was not declared in this scope
   lua_pushglobaltable(L);         // S: ... string globals

I tried to delele "Werror" in lua_parameter_dictionary.cc and compile again but it didn't work.....

how can I fix it? Thanks a lot!

damonkohler, 2016-10-28:
Please provide some information about your system (run the commands below) and copy the full compilation log to a gist.

```
g++ -v
uname -a
lsb_release -a
```

ywj447, 2016-10-28:
ubuntu@tegra-ubuntu:~$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 4.8.4-2ubuntu1~14.04' --with-bugurl=file:///usr/share/doc/gcc-4.8/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.8 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.8 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libmudflap --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.8-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.8-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.8-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.8.4 (Ubuntu/Linaro 4.8.4-2ubuntu1~14.04)

ubuntu@tegra-ubuntu:~$ uname -a
Linux tegra-ubuntu 3.10.40-grinch-21.2.1 #8 SMP PREEMPT Sat Dec 13 11:27:12 UTC 2014 armv7l armv7l armv7l GNU/Linux

ubuntu@tegra-ubuntu:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 14.04.1 LTS
Release:        14.04
Codename:       trusty

thanks @damonkohler

SirVer, 2016-10-28:
Please also provide output of `dpkg -l` - it seems that something is screwy with your Lua installation.

ywj447, 2016-10-31:
ii  libdaemon0     0.14-2ubuntu armhf        lightweight C library for daemons
ii  libdatrie1:arm 0.2.8-1      armhf        Double-array trie library
ii  libdb5.3:armhf 5.3.28-3ubun armhf        Berkeley v5.3 Database Libraries
ii  libdbus-1-3:ar 1.6.18-0ubun armhf        simple interprocess messaging sys
ii  libdbus-1-dev: 1.6.18-0ubun armhf        simple interprocess messaging sys
ii  libdbus-cpp2:a 2.0.0+14.04. armhf        header-only dbus-binding leveragi
ii  libdbus-glib-1 0.100.2-1    armhf        simple interprocess messaging sys
ii  libdbusmenu-gl 12.10.3+14.0 armhf        library for passing menus over DB
ii  libdbusmenu-gt 12.10.3+14.0 armhf        library for passing menus over DB
ii  libdbusmenu-gt 12.10.3+14.0 armhf        library for passing menus over DB
ii  libdbusmenu-qt 0.9.3+14.04. armhf        Qt implementation of the DBusMenu
ii  libdbusmenu-qt 0.9.3+14.04. armhf        Qt5 implementation of the DBusMen
ii  libdc1394-22:a 2.2.1-2ubunt armhf        high level programming interface
ii  libdc1394-22-d 2.2.1-2ubunt armhf        high level programming interface
ii  libdca0:armhf  0.0.5-6ubunt armhf        decoding library for DTS Coherent
ii  libdconf1:armh 0.20.0-1     armhf        simple configuration storage syst
ii  libdebconfclie 0.187ubuntu1 armhf        Debian Configuration Management S
rc  libdebian-inst 0.88ubuntu4  armhf        Library of common debian-installe
ii  libdecoration0 1:0.9.11.2+1 armhf        Compiz window decoration library
ii  libdee-1.0-4:a 1.2.7+14.04. armhf        model to synchronize multiple ins
rc  libdevmapper-e 2:1.02.77-6u armhf        Linux Kernel Device Mapper event
ii  libdevmapper1. 2:1.02.77-6u armhf        Linux Kernel Device Mapper usersp
ii  libdigest-hmac 1.03+dfsg-1  all          module for creating standard mess
ii  libdirac-encod 1.0.2-6ubunt armhf        open and royalty free high qualit
ii  libdirectfb-1. 1.2.10.0-5   armhf        direct frame buffer graphics - sh
ii  libdjvulibre-t 3.5.25.4-3   all          Linguistic support files for libd
ii  libdjvulibre21 3.5.25.4-3   armhf        Runtime support for the DjVu imag
ii  libdmapsharing 2.9.24-0ubun armhf        DMAP client and server library -
rc  libdmraid1.0.0 1.0.0.rc16-4 armhf        Device-Mapper Software RAID suppo
ii  libdns100      1:9.9.5.dfsg armhf        DNS Shared Library used by BIND
ii  libdotconf0:ar 1.3-0ubuntu2 armhf        Configuration file parser library
ii  libdpkg-perl   1.17.5ubuntu all          Dpkg perl modules
ii  libdrm-dev:arm 2.4.60-2~ubu armhf        Userspace interface to kernel DRM
ii  libdrm-exynos1 2.4.60-2~ubu armhf        Userspace interface to exynos-spe
ii  libdrm-freedre 2.4.60-2~ubu armhf        Userspace interface to msm/kgsl k
ii  libdrm-nouveau 2.4.60-2~ubu armhf        Userspace interface to nouveau-sp
ii  libdrm-omap1:a 2.4.60-2~ubu armhf        Userspace interface to omap-speci
ii  libdrm-radeon1 2.4.60-2~ubu armhf        Userspace interface to radeon-spe
ii  libdrm2:armhf  2.4.60-2~ubu armhf        Userspace interface to kernel DRM
ii  libdv4:armhf   1.0.0-6      armhf        software library for DV format di
ii  libdvdnav4:arm 4.2.1-3      armhf        DVD navigation library
ii  libdvdread4:ar 4.2.1-2ubunt armhf        library for reading DVDs
ii  libebackend-1. 3.10.4-0ubun armhf        Utility library for evolution dat
ii  libebook-1.2-1 3.10.4-0ubun armhf        Client library for evolution addr
ii  libebook-conta 3.10.4-0ubun armhf        Client library for evolution cont
ii  libecal-1.2-16 3.10.4-0ubun armhf        Client library for evolution cale
ii  libecryptfs0   104-0ubuntu1 armhf        ecryptfs cryptographic filesystem
ii  libedata-book- 3.10.4-0ubun armhf        Backend library for evolution add
ii  libedata-cal-1 3.10.4-0ubun armhf        Backend library for evolution cal
ii  libedataserver 3.10.4-0ubun armhf        Utility library for evolution dat
ii  libedit2:armhf 3.1-20130712 armhf        BSD editline and history librarie
ii  libegl1-mesa:a 10.1.3-0ubun armhf        free implementation of the EGL AP
ii  libegl1-mesa-d 10.1.3-0ubun armhf        free implementation of the EGL AP
ii  libegl1-mesa-d 10.1.3-0ubun armhf        free implementation of the EGL AP
ii  libeigen3-dev  3.2.0-8      all          lightweight C++ template library
ii  libelf1:armhf  0.158-0ubunt armhf        library to read and write ELF fil
ii  libelfg0:armhf 0.8.13-5     armhf        an ELF object file access library
ii  libemail-valid 1.192-1      all          Perl module for checking the vali
ii  libenca0:armhf 1.15-2       armhf        Extremely Naive Charset Analyser
ii  libenchant1c2a 1.6.0-10ubun armhf        Wrapper library for various spell
ii  liberror-perl  0.17-1.1     all          Perl module for error/exception h
ii  libespeak1:arm 1.47.11-1ubu armhf        Multi-lingual software speech syn
ii  libestr0       0.1.9-0ubunt armhf        Helper functions for handling str
ii  libevdocument3 3.10.3-0ubun armhf        Document (PostScript, PDF) render
ii  libevemu1:armh 1.2.0-1ubunt armhf        Linux Input Event Device Emulatio
ii  libevent-2.0-5 2.0.21-stabl armhf        Asynchronous event notification l
ii  libevview3-3   3.10.3-0ubun armhf        Document (PostScript, PDF) render
ii  libexempi3:arm 2.2.1-1ubunt armhf        library to parse XMP metadata (Li
ii  libexif-dev    0.6.21-1ubun armhf        library to parse EXIF files (deve
ii  libexif12:armh 0.6.21-1ubun armhf        library to parse EXIF files
[... 1959 lines cut ...]

[... cut ...]
ii  activity-log-manager-control-center                   0.9.7-0ubuntu14                                     all          blacklist configuration for Zeitgeist (transitional package)
ii  adduser                                               3.113+nmu3ubuntu3                                   all          add and remove users and groups
ii  adium-theme-ubuntu                                    0.3.4-0ubuntu1                                      all          Adium message style for Ubuntu

@SirVer  here is the output of dpkg -l.  thanks you very much!

SirVer, 2016-10-31:
Thank you. The team is traveling this week, so I will not have time to look at this until next week.

ywj447, 2016-10-31:
I appreciate  you and your team's generous help @SirVer 
it's OK. I am trying to fix it and looking forward  to your reply.
Have a nice trip！

SirVer, 2016-11-14:
First, you have lua-5.1 installed - that will probably cause issues. Second - and sorry for not seeing this earlier - you have a ARM processor. We never tried to run Cartographer on ARM - and that is also unsupported. We know for a fact though that it requires 64bit and does not work with 32bit. Is your arm architecture 64bit?

damonkohler, 2016-11-14:
I'm not sure if 64-bit matters on ARM. Our 64-bit requirement is about Eigen alignment/vectorization.

ywj447, 2016-11-15:
@SirVer @damonkohler  thanks for your help!
I have solved this problem and run cartographer successfully after I had reinstalled all the ROS.

closed 2016-11-15 by damonkohler