cartographer-project/cartographer_ros #158: Add support for ROS versions Jade and Kinetic

damonkohler, 2016-10-31:
Likely this only requires additional Travis configurations.

ruffsl, 2016-12-07:
I'm getting a `collect2 error ld returned 1 exit status` error when building with kinetic,
simalur to this: https://github.com/googlecartographer/cartographer_ros/issues/159#issuecomment-260423459
Log output here: http://pastebin.com/qbyCKFDi
The dockerfile that generated the log above can be found [here](
https://github.com/ruffsl/cartographer_ros/blob/2d46e7b0272f280cc345ed4cfca8a69f5b072833/docker/Dockerfile).

SirVer, 2017-06-08:
This requires new Dockerfiles and changes to travis.yml to make sure that it runs on CI and we do not regress in the support.

mikaelarguedas, 2017-08-11:
The back-port of ceres-solver 1.12.0 for all Lunar targeted platforms (Ubuntu X,Y and Z) is now complete and the packages are available at packages.ros.org. So all the cartographer dependencies are now available for ROS Lunar.
@damonkohler and @jack-oquin what would be the way forward to make an initial release of `cartographer`/`cartographer_ros` in ROS Lunar? I guess targeting the last source release 0.2.0 is preferred?

damonkohler, 2017-08-11:
We do not have a regular schedule for doing source releases, so 0.2.0 is a reasonable place to start.

mikaelarguedas, 2017-08-11:
thanks for the quick answer, I'll wait for @jack-oquin's feedback regarding release details (where to host the release repository and who should receive emails in case of failure to build the binaries, etc)

mikaelarguedas, 2017-08-21:
@jack-oquin friendly ping.
I'm happy to give a hand for the release. The current unknowns being:
- where should we host the release repository? (I guess the most convenient location would be on [ros-gbp](https://github.com/ros-gbp) alongside the other release repos)
- who should be notified on build failures? (if we leave the package.xml as is, the google-cartographer mailing list will receive the buildfarm emails, I don't know if this is desired or should be avoided)

Cheers!

mikaelarguedas, 2017-09-07:
I'm going to go ahead and create the release repositories on ros-gbp.

> who should be notified on build failures? (if we leave the package.xml as is, the google-cartographer mailing list will receive the buildfarm emails, I don't know if this is desired or should be avoided)

@damonkohler could you give some feedback on this? I don't want the mailing list to be spammed if it's not of interest for the cartographer developers

Thanks!

SirVer, 2017-10-12:
@mikaelarguedas We should probably create a new google group for emails as such, so only devs receive these emails. @cschuet time for a cartographer-ninja@ or cartographer-owners@ mailing list?

mikaelarguedas, 2017-10-18:
> We should probably create a new google group for emails as such, so only devs receive these emails.

Sounds good to me :+1: 

I held the Lunar release for now because building debs based on v2.0.0 fails. I saw that @jspricke made [some changes](https://github.com/googlecartographer/cartographer/commit/e345ae8280e9f595ba21432288a33d323e2e9fcd) to that effect after 2.0.0 was released, so maybe the current state of master would build successfully.

cschuet, 2017-10-18:
I created cartographer-owners@googlegroups.com for this purpose.

mikaelarguedas, 2017-10-18:
awesome thanks @cschuet !

k-okada, 2017-11-01:
Hi, since ceres-solvers has been released under http://packages.ros.org/ros/ubuntu/pool/main/c/ceres-solver/ ( https://discourse.ros.org/t/ceres-package/872). Can we release cartographer form ros buildfarm?
;; Since cartographer-owners@googlegroups.com seems private, so posted here.

mikaelarguedas, 2017-11-01:
Hi @k-okada, indeed cartographer now has all its dependencies available to be released in ROS Lunar. I tried to release 0.2.0 in Lunar about a month ago but it failed so I reverted it https://github.com/ros/rosdistro/pull/15984.

@damonkohler @SirVer Is there a release date planned for 0.3.0 ? Or should I give a shot to releasing the current state of master ?

@cschuet I'll make a PR to this package and cartographer_ros to make cartographer-owners@googlegroups.com the maintainer's email address. Is it possible to add "noreply@build.ros.org" to that google group so that the buildfarm can send emails to it ? thanks!

SirVer, 2017-11-14:
@damonkohler @cschuet Could you guys cut a 0.3 release for this?

mikaelarguedas, 2017-12-17:
Thanks for cutting 0.3.0.
I gave another shot to release this into Lunar, but as some dependencies changed since 0.2.0, it can't be released as is.

Dependency changes since 0.2.0:
- ceres 1.13: need to redo the backporting for all distros (as for 1.12 a few months ago)
- protobuf 3 : not available in xenial, 3.0.0 available in Zesty and Debian Stretch. It built fine with 3.0.0 but please let me know if you're aware of a higher minimum version of protobuf.

cschuet, 2017-12-18:
Thanks for taking a stab at this again and sorry for the dependency changes. Let me add some context.

**ceres 1.12 => 1.13**: We talked to the Ceres team and learned from them that they were able to make a couple of performance improving changes to their library and suggested to move to the newer version. We found that we are getting some 5% drop in wall time by just moving to 1.13. So that was irresistible :)

**protobuf 2 => 3**: The reason why we had to upgrade to protobuf 3 is [cloud-based mapping](https://github.com/googlecartographer/rfcs/blob/master/text/0002-cloud-based-mapping-1.md) as we need to send sensor data and intermediate mapping results over the wire and want to use gRPC for that which only supports proto3. This code lives in the cartographer_grpc directory and compilation is hidden behind the `BUILD_GRPC=ON` compilation flag. If you don't care for that code, there is no requirement on the minimum protobuf 3 version, so 3.0.0 is absolutely fine.

mikaelarguedas, 2017-12-18:
Thanks @cschuet for the additional context.
As the released version of ceres is broken on Ubuntu I think it's fine to repackage and distribute our own 1.13. For protobuf that's more problematic as I expect other projects to rely on the system version and would rather not package a version different from upstream with potential side effect on other ROS packages or parts of the system.

Maybe I'll just hold this for another few months and release cartographer only in the [upcoming ROS-Melodic](https://github.com/ros-infrastructure/rep/pull/139) that targets Ubuntu Bionic and Debian Stretch that both have all the dependencies versions cartographer 0.3.0 uses.

cschuet, 2018-03-10:
Closing. Lunar seems not feasible, other versions are supported.

closed 2018-03-10 by cschuet

cschuet, 2018-03-10:
Closing as kinetic and lunar are now supported and jade is EOL.

mikaelarguedas, 2018-03-19:
@cschuet Re "mailing list and ROS build farm notifications": We are about to release cartographer in various ROS distributions (0.2 into Kinetic and Lunar and 0.3 into upcoming Melodic). Can you please confirm that "noreply@build.ros.org" is allowed to post messages to the cartographer-owners@googlegroups.com mailing list ? (as I'm not a member of that group I cannot confirm it by myself)

Thanks!

cschuet, 2018-03-19:
@mikaelarguedas Thanks for the heads up. noreply@build.ros.org should be able to email to the cartographer-owners@googlegroups.com mailing list