cartographer-project/cartographer_ros #159: Running error

Needrom, 2016-11-01:
It will have an error when I run the backpack_2d.launch or demo_revo_2d.launch if the data of laserscan transmit to the cartographer_node

Needrom, 2016-11-01:
Demo_revo_lds.launch not demo_revo_2d.launch

SirVer, 2016-11-01:
Please paste the full output and the commandline you are running. 

Also, we will likely be slow to responsd since the team is currently traveling.

Needrom, 2016-11-02:
thanks here's the error info
[FATAL] [1477735379.346794350]: F1029 03:02:59.000000 14494 sensor_data.cc:34] Check failed: frame_id[0] != '/' ('/' vs. '/') 
*** Check failure stack trace: ***
    @     0x7fa26cf0cdaa  (unknown)
    @     0x7fa26cf0cce4  (unknown)
    @     0x7fa26cf0c6e6  (unknown)
    @     0x7fa26cf0f687  (unknown)
    @           0x5107f0  cartographer_ros::SensorData::SensorData()
    @           0x512e54  cartographer::common::make_unique<>()
    @           0x511317  cartographer_ros::SensorDataProducer::AddLaserScanMessage()
    @           0x4e59d8  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros12_GLOBAL__N_14Node10InitializeEvEUlRKNS_10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEEE_vSE_E6invokeERNS1_15function_bufferESE_
    @           0x4f1ccb  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5042b8  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7fa26e2766b5  (unknown)
    @     0x7fa26e230107  (unknown)
    @     0x7fa26e230c33  (unknown)
    @     0x7fa26e2791e5  (unknown)
    @     0x7fa26e260e0b  (unknown)
    @           0x4eacae  cartographer_ros::(anonymous namespace)::Run()
    @           0x4e41dd  main
    @     0x7fa26b1c9f45  (unknown)
    @           0x4e5327  (unknown)
    @              (nil)  (unknown)
[cartographer_node-2] process has died [pid 14494, exit code -6, cmd /home/liangguoyu/catkin_ws/devel/lib/cartographer_ros/cartographer_node -configuration_directory /home/liangguoyu/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files -configuration_basename backpack_2d.lua echoes:=horizontal_laser_2d __name:=cartographer_node __log:=/home/liangguoyu/.ros/log/365f183e-9d11-11e6-90d6-000c29931a89/cartographer_node-2.log].
log file: /home/liangguoyu/.ros/log/365f183e-9d11-11e6-90d6-000c29931a89/cartographer_node-2*.log

SirVer, 2016-11-02:
Your frame_ids start with a `/`. This is deprecated behavior and because of inconsistent handling in various ROS packages, we do not support that. Remove the slashes and it should work.

Needrom, 2016-11-02:
En I found an doc and said just like you.  I think I should write an .launch or .py to convert it. Do the ROS have that message type or setting so I can skip this step.

Needrom, 2016-11-03:
how to reopen .can't find that option or notice

SirVer, 2016-11-03:
> En I found an doc and said just like you. I think I should write an
> .launch or .py to convert it. Do the ROS have that message type or setting
> so I can skip this step.

​I do not understand the question. Do you mean how to remove the leading
slashes in your data? I would write a python script using the python rosbag
API. ​

Google Germany GmbH
Erika-Mann-Straße 33
80331 München

Registergericht und -nummer: Hamburg, HRB 86891
Sitz der Gesellschaft: Hamburg
Geschäftsführer: Matthew Scott Sucherman, Paul Terence Manicle

Needrom, 2016-11-04:
thanks.  I finished it. 

but I still no map message . I think that because my odom setting is no right .

and then I found what the demo_2d.bag provide is Imu and Laserscan message . But actually I do not have Imu .I only have odom. Could it be work with only Laserscan and odom ?

at the backpack_2d.lua . I found this three setting and it's all setting  false when running the demo
` provide_odom_frame = false,
`  
`use_odometry_data = false,
`  
`use_constant_odometry_variance = false,
`

sorry for asking in an old issue .

SirVer, 2016-11-04:
There is a [setting about expecting imu](https://github.com/googlecartographer/cartographer/blob/master/configuration_files/trajectory_builder_2d.lua#L16). You can set this to false and then only rely on scan matching. You will want to turn on the [correlative scan matcher](https://github.com/googlecartographer/cartographer/blob/master/configuration_files/trajectory_builder_2d.lua#L21), which is slower but tries harder to find a match.

You can throw in odometry then, but I would try to get scan matching working first.

Needrom, 2016-11-05:
here's the whole settings  ,I already set the imu to false and only_scan_matching to true .But what I see in the rviz is stilling nothing on the map.And info tell me the map is fade. 

-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "/odom",     --/odom  /moblie_base/moblie_base_controller/odom
  provide_odom_frame = false,
  use_odometry_data = false,
  use_constant_odometry_variance = false,
  constant_odometry_translational_variance = 0.,
  constant_odometry_rotational_variance = 0.,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.,
  horizontal_laser_max_range = 30.,
  horizontal_laser_missing_echo_ray_length = 5.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

return options

                                                             
here's the info which is full of terminal.
`[ WARN] [1478334791.211842720]: Could not compute submap fading: "map" passed to lookupTransform argument target_frame does not exist. 
`

Needrom, 2016-11-07:
Sorry for providing the Incomplete information. I had many try in the past two day. Nothing work . 

[image]

and here is the command line for rviz 

[ INFO] [1478507022.617061829]: rviz version 1.11.14
[ INFO] [1478507022.617236191]: compiled against Qt version 4.8.6
[ INFO] [1478507022.617306819]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1478507023.057659224]: Stereo is NOT SUPPORTED
[ INFO] [1478507023.057827988]: OpenGl version: 3 (GLSL 1.3).
[ WARN] [1478507039.073142005]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1478507099.098805524]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.
[ WARN] [1478507159.127116399]: MessageFilter [target=map ]: Dropped 100.00% of messages so far. Please turn the [ros.rviz.message_notifier] rosconsole logger to DEBUG for more information.

 rqt_graph

[image]

here is the roslaunch 

<launch>
  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/backpack_2d.urdf" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

<node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename backpack_2d.lua"
      output="screen">
    <remap from="echoes" to="horizontal_laser_2d" />
  </node>
</launch>

here is the backpack_2d.lua
- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "/mobile_base/mobile_base_controller/odom",     --/odom  /moblie_base/moblie_base_controller/odom
  provide_odom_frame = false,
  use_odometry_data = false,
  use_constant_odometry_variance = false,
  constant_odometry_translational_variance = 0.,
  constant_odometry_rotational_variance = 0.,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.,
  horizontal_laser_max_range = 30.,
  horizontal_laser_missing_echo_ray_length = 5.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

return options

SirVer, 2016-11-08:
Your rviz output shows that `/map` is not published, i.e. Cartographer is not SLAMing. You did not provide its output though, so this is still very hard to debug. The console output of the `roslaunch` command you are using and the log files written by Cartographer are required for that.

SirVer, 2016-11-08:
One theory: I see that you remap from `echoes` to `horizontal_laser_2d` - which sounds like you have a multi echo laser. In your lua file you set `use_horizontal_multi_echo_laser` to false though and instead expect the non-multi-echo laser. Could that be the issue?

Needrom, 2016-11-08:
thanks all.
But I could't use the type of `horizontal_multi_echo_laser`.Because I don't have  radar,I am using the camera of asus xtion pro live.I get the laser_scan through the pkg of 'depthimage_to_laserscan'.This pkg can only convert the depthimage message to laserscan. 
    it won't recieve my '/scan' topic when I set the `use_horizontal_multi_echo_laser` to true . And the roslaunch won't let me to set them both to be true, it will output an error.
   This so much confusion now.but i saw it efficiency on the youtube,I want to let it work so much.I use to slam by the `gmapping` but it did't meet my expectation.
   if any setting I miss or have some way to debug what's wrong. tell me.Good to receive your help.

SirVer, 2016-11-08:
Have you tried changing `echoes` to `scan` in the line below? Imho that is the topic `gmapping` is expecting, so if you did not change it, you have to let cartographer know that you are publishing your lasers there.

~~~
    <remap from="echoes" to="horizontal_laser_2d" />
~~~

Otherwise you really have to upload a git repo somewhere that contains all your launch files. It is clear that your scans do not reach Cartographer - as of why is not identifiable with the information you provided so far.

Needrom, 2016-11-12:
I got an radar .No more error about the map on the rviz .But still no any map show on screan
[image]

terminal info .it's because of the rate?
```
WARNING: Package name "firstPackage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
... logging to /home/liangguoyu/.ros/log/bc1bb888-a874-11e6-bb2e-000c29931a89/roslaunch-ubuntu-9631.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "firstPackage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
WARNING: Package name "firstPackage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
started roslaunch server http://ubuntu:46031/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Package name "firstPackage" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[cartographer_node-1]: started with pid [9649]
process[rviz-2]: started with pid [9650]
[ INFO] [1478918383.560418880]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/home/liangguoyu/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/revo_lds.lua' for 'revo_lds.lua'.
[ INFO] [1478918383.560938447]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1478918383.561342041]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1478918383.561542305]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1478918383.561650866]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1478918383.561864639]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1478918383.561979532]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1478918383.562434197]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1478918383.563824895]: I1111 18:39:43.000000  9649 configuration_file_resolver.cc:40] Found '/usr/local/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1478918383.739233430]: I1111 18:39:43.000000  9649 submaps.cc:176] Added submap 1
[ INFO] [1478918383.922044668]: I1111 18:39:43.000000  9649 cartographer_node_main.cc:615] neato_laser rate: unknown
```

here is the launch I use 
```
<launch>
  <param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"  type="cartographer_node" args="-configuration_directory $(find cartographer_ros)/configuration_files  -configuration_basename revo_lds.lua"  output="screen">
    <remap from="scan" to="/scan" />
  </node>
  <node name="rviz" pkg="rviz" type="rviz" required="true" args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
</launch>
```

and here is the revo_lds.lua .without any change
```
include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "horizontal_laser_link",
  published_frame = "horizontal_laser_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_constant_odometry_variance = false,
  constant_odometry_translational_variance = 0.,
  constant_odometry_rotational_variance = 0.,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.3,
  horizontal_laser_max_range = 8.,
  horizontal_laser_missing_echo_ray_length = 1.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options
```

Needrom, 2016-11-14:
thanks   
 I start to rebuild the workspace ,because some option I set it doesn't work actually. such as I set the  imu false it actually work on the tf tree . 
   the source I use wasn't the original version write by google's team .It's a modified version by some else saying it have fewer setting on it . I don't want to doubt it,because the demo work well. it possible due to my wrong operation.but I still want to rebuild the space.
   this time I follow all setting from cartographer wiki,but not some one blog.And get a problem I never seen.

when I run `catkin_make_isolated --install --use-ninja`.
the error  occurs.
```
FAILED: /usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -Wno-deprecated -O3 -DNDEBUG -isystem /usr/include/vtk-5.8 -isystem /usr/include/eigen3 -isystem /home/liangguoyu/catkin_for_cartographer/install_isolated/include -isystem /opt/ros/indigo/include -isystem /usr/include/pcl-1.7 -isystem /usr/include/lua5.2 -I. -I/home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros -I/home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros -isystem /usr/include/ni     -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o -MF "cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o.d" -o cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o -c /home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc
/home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc: In member function ‘bool cartographer_ros::{anonymous}::Node::HandleSubmapQuery(cartographer_ros_msgs::SubmapQuery::Request&, cartographer_ros_msgs::SubmapQuery::Response&)’:
/home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros/cartographer_ros/node_main.cc:282:38: error: ‘cartographer_ros_msgs::SubmapQuery::Request’ has no member named ‘submap_index’
       request.trajectory_id, request.submap_index, &response_proto);
                                      ^
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh', 'ninja', '-j1', '-l1']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros && /home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh ninja -j1 -l1

Command failed, exiting.

```

And I try to run `./env.sh ninja -j1 -l1` in `~/catkin_for_cartographer/install_isolated` directly.
this error occurs
`ninja: error: loading 'build.ninja': No such file or directory`

SirVer, 2016-11-14:
Can you link us this blog post? Having less settings would certainly be nice! :)

It seems you did not git pull on cartographer_ros. Did you update all of your git repositories to latest head?

Needrom, 2016-11-14:
I should be already install googletest why I still have that error

SirVer, 2016-11-14:
> I should be already install googletest why I still have that error

I do not understand that question. You have a compile error because cartographer and cartographer_ros are not in sync on your system. You probably have to git pull on all ROS repos.

Thanks for the links! The fork you linked seems to only have checked in some new documentation and temporary files and removed some other documentation. No functional changes that I can see.

Needrom, 2016-11-14:
en I actually git clone the master url of your not by my own repos(  [https://github.com/googlecartographer/cartographer_ros.git](url)  )
it should be the latest

SirVer, 2016-11-14:
In your last post there is very little information, for example the output of what step failed in the configuration is missing. Please post the full output.

Needrom, 2016-11-14:
I give you this first .It have some command i try directly. I give the whole `catkin_make_isolated --install --use-ninja` output later . 

```
/mapping_3d/motion_filter.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/optimizing_local_trajectory_builder.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/optimizing_local_trajectory_builder_options.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/rotation_cost_function.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/sparse_pose_graph.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/submaps.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/translation_cost_function.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/kalman_local_trajectory_builder_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/laser_fan_inserter_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/local_trajectory_builder_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/motion_filter_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/optimizing_local_trajectory_builder_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/proto/submaps_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/ceres_scan_matcher.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/fast_correlative_scan_matcher.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/interpolated_grid.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/occupied_space_cost_functor.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/precomputation_grid.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/real_time_correlative_scan_matcher.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/rotation_delta_cost_functor.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/rotational_scan_matcher.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/translation_delta_cost_functor.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/proto/ceres_scan_matcher_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/scan_matching/proto/fast_correlative_scan_matcher_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/sparse_pose_graph/constraint_builder.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/sparse_pose_graph/optimization_problem.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/mapping_3d/sparse_pose_graph/spa_cost_function.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/collator.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/compressed_point_cloud.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/configuration.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/data.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/laser.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/ordered_multi_queue.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/point_cloud.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/voxel_filter.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/proto/adaptive_voxel_filter_options.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/proto/configuration.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/sensor/proto/sensor.pb.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/transform/rigid_transform.h
-- Installing: /home/liangguoyu/catkin_for_cartographer/install_isolated/include/cartographer/transform/rigid_transform_test_helpers.h
[... cut ...] erialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
-- Configuring incomplete, errors occurred!
See also "/home/liangguoyu/catkin_for_cartographer/CMakeFiles/CMakeOutput.log".
See also "/home/liangguoyu/catkin_for_cartographer/CMakeFiles/CMakeError.log".

```

Needrom, 2016-11-14:
here is the whole output
```
[1/8] Creating directories for 'ceres_src'
[2/8] Performing download step (git clone) for 'ceres_src'
ninja: build stopped: interrupted by user.
Base path: /home/liangguoyu/catkin_for_cartographer
Source space: /home/liangguoyu/catkin_for_cartographer/src
Build space: /home/liangguoyu/catkin_for_cartographer/build_isolated
Devel space: /home/liangguoyu/catkin_for_cartographer/devel_isolated
Install space: /home/liangguoyu/catkin_for_cartographer/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - ceres_solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing plain cmake package: 'ceres_solver'
==> Creating build directory: 'build_isolated/ceres_solver/install'
==> cmake /home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/ceres_solver -DCMAKE_INSTALL_PREFIX=/home/liangguoyu/catkin_for_cartographer/install_isolated -G Ninja in '/home/liangguoyu/catkin_for_cartographer/build_isolated/ceres_solver/install'
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Found Git: /usr/bin/git (found version "1.9.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/liangguoyu/catkin_for_cartographer/build_isolated/ceres_solver/install
==> ninja -j1 -l1 in '/home/liangguoyu/catkin_for_cartographer/build_isolated/ceres_solver/install'
[1/7] Performing download step (git clone) for 'ceres_src'
Base path: /home/liangguoyu/catkin_for_cartographer
Source space: /home/liangguoyu/catkin_for_cartographer/src
Build space: /home/liangguoyu/catkin_for_cartographer/build_isolated
Devel space: /home/liangguoyu/catkin_for_cartographer/devel_isolated
Install space: /home/liangguoyu/catkin_for_cartographer/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - ceres_solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros_msgs
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing plain cmake package: 'ceres_solver'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/liangguoyu/catkin_for_cartographer/build_isolated/ceres_solver/install'
ninja: no work to do.
==> ninja -j1 -l1 in '/home/liangguoyu/catkin_for_cartographer/build_isolated/ceres_solver/install'
ninja: build stopped: interrupted by user.
[1/7] Performing download step (git clone) for 'ceres_src'
Cloning into 'ceres_src'...
Note: checking out '6a13e39e8171f450fbb89188d97f198def81937e'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

[... 1119 lines cut ...]
--   package 'openni-dev' not found
-- Found openni: /usr/lib/libOpenNI.so  
-- Found libusb-1.0: /usr/include  
-- looking for PCL_COMMON
-- Found PCL_COMMON: /usr/lib/libpcl_common.so  
-- looking for PCL_OCTREE
-- Found PCL_OCTREE: /usr/lib/libpcl_octree.so  
-- looking for PCL_IO
-- Found PCL_IO: /usr/lib/libpcl_io.so  
-- Found PCL: /usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;optimized;/usr/lib/libpcl_common.so;debug;/usr/lib/libpcl_common.so;optimized;/usr/lib/libpcl_octree.so;debug;/usr/lib/libpcl_octree.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts;optimized;/usr/lib/libpcl_io.so;debug;/usr/lib/libpcl_io.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_iostreams.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libpthread.so;/usr/lib/libOpenNI.so;vtkCommon;vtkRendering;vtkHybrid;vtkCharts  
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   iostreams
-- Configuring incomplete, errors occurred!
See also "/home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
[... cut ...] , 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros && /home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh cmake /home/liangguoyu/catkin_for_cartographer/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/liangguoyu/catkin_for_cartographer/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/liangguoyu/catkin_for_cartographer/install_isolated -G Ninja

```

Needrom, 2016-11-14:
correct the from [ http://www.cnblogs.com/yebo92/p/5940856.html](url)   to  [http://www.cnblogs.com/hitcm/p/5939507.html](url)

Needrom, 2016-11-14:
I read the CMakeError.log
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/ninja cmTryCompileExec3956915166
[1/2] Building C object CMakeFiles/cmTryCompileExec3956915166.dir/CheckSymbolExists.c.o
[2/2] Linking C executable cmTryCompileExec3956915166
FAILED: : && /usr/bin/cc     CMakeFiles/cmTryCompileExec3956915166.dir/CheckSymbolExists.c.o  -o cmTryCompileExec3956915166  -rdynamic && :
CMakeFiles/cmTryCompileExec3956915166.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

File /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros/CMakeFiles/CMakeTmp

Run Build Command:/usr/bin/ninja cmTryCompileExec1816991125
[1/2] Building C object CMakeFiles/cmTryCompileExec1816991125.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTryCompileExec1816991125
FAILED: : && /usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTryCompileExec1816991125.dir/CheckFunctionExists.c.o  -o cmTryCompileExec1816991125  -rdynamic -lpthreads && :
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

it tell me CheckFunctionExists.c failed in function pthread_create . but why I cannot find the .c in the path the log said.

`/usr/bin/ld: cannot find -lpthreads` .I just can find libpthread.so .Whether It should write `-lpthread` .

SirVer, 2016-11-15:
Thanks, that gets us a bit closer. Could you please provide the output of `lsb_release -a`? 

The pthread issue is fine - that is just a try compile. and the rest of the output of cmake shows that it settled on using `-lpthread`. The problem is really gtest - could you try commenting the `google_catkin_test` in `cartographer_ros/CMakeLists.txt` and try again?

Needrom, 2016-11-17:
`lsb_release -a`output 
```
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty
```

I'm already comment `google_enable_testing()` .cannot find the `google_catkin_test` .Am I commenting the right place? I still get the same error from `catkin_make_isolated --install --use-ninja`.

SirVer, 2016-11-17:
The good news is we are now done to some conflict between catkin_tests and regular gunit. Not sure what is going on though, I cannot reproduce this with Ubuntu 14.04 in a docker image. 

Here is the `google_catkin_test` that you could try to comment out  `https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/CMakeLists.txt#L103`.

Needrom, 2016-11-18:
even though the compile was still error. thanks you any way.
```
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros': 
  Command '['/home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh', 'ninja', '-j1', '-l1']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer_ros && /home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh ninja -j1 -l1

Command failed, exiting.
```
this error will be occure while the `google_catkin_test` to be comment and `google_enable_testing()` for not.

if both of them to be comment.
```
-- Configuring incomplete, errors occurred!
See also "/home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer/install/CMakeFiles/CMakeOutput.log".
See also "/home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer/install/CMakeFiles/CMakeError.log".
ninja: error: rebuilding 'build.ninja': subcommand failed
<== Failed to process package 'cartographer': 
  Command '['/home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh', 'ninja', 'build.ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/liangguoyu/catkin_for_cartographer/build_isolated/cartographer && /home/liangguoyu/catkin_for_cartographer/install_isolated/env.sh ninja build.ninja

Command failed, exiting.
```

the other message is similar whether you want the whole output.

Needrom, 2016-11-18:
having a question . actually I already build cartographer though here
```
cd cartographer
mkdir build
cd build
cmake .. -G Ninja
ninja
ninja test
sudo ninja install
```
that is what call the Cartographer’s system requirements.but the file is similar to `catkin_ws/build_isolated/cartographer/install` which I compile all the time.

Needrom, 2016-11-19:
thanks you so much for helping me such a long time!!! I already let it work with the source of hitcm .But the source I clone from google master still get error from `ninja build.ninja`.I have no idea sorry.

 I reinstall all the package and remake the source .Add the tf from `base_link` to `laser` .The rviz show me a map from my location. 

Two problem may cause no map I think.The first is the souce may not be made properly.Second may blame my laser. It rate is too law .When I set the `tf`  rate to 10hz the map become to stable.And when the `tf` rate raise up to 20hz nothing on the rviz.

closed 2016-11-21 by Needrom

SirVer, 2016-11-21:
Yes, you should be able to replace your cartographer through the one in your home simply through symlinking. but the versions might no longer be compatible. 

Please always post full output of your error logs, otherwise it is really hard to understand what you changed and what the current status is. 

If your laser has a low rate, you will need to adjust `https://github.com/googlecartographer/cartographer/blob/master/configuration_files/trajectory_builder_3d.lua#L20` - depending on your configuration this value might have a value > 1.

Needrom, 2016-11-22:
en OK I try 3dSLAM later.but does cartographer have Navigation ? Or I have to use other algorithm.

Maxinjun, 2020-04-16:
> thanks. I finished it.
> 
> but I still no map message . I think that because my odom is not setting right .
> 
> and then I found what the demo_2d.bag provide is Imu and Laserscan message . But actually I do not have Imu .I only have odom. Could it be work with only Laserscan and odom ?
> 
> at the backpack_2d.lua . I found this three setting and it's all setting false when running the demo
> `provide_odom_frame = false, `
> `use_odometry_data = false, `
> `use_constant_odometry_variance = false, `
> 
> could I use odom instead of the imu ?

> thanks. I finished it.
> 
> but I still no map message . I think that because my odom is not setting right .
> 
> and then I found what the demo_2d.bag provide is Imu and Laserscan message . But actually I do not have Imu .I only have odom. Could it be work with only Laserscan and odom ?
> 
> at the backpack_2d.lua . I found this three setting and it's all setting false when running the demo
> `provide_odom_frame = false, `
> `use_odometry_data = false, `
> `use_constant_odometry_variance = false, `
> 
> could I use odom instead of the imu ?

@Needrom 
  Hi,
I have the same issue, need to remove '/'，I am a newer for ros ,Could you share me how to romove it? Thanks a lot