cartographer-project/cartographer_ros #161: Build error on catkin_make_isolated --install --use-ninja

etsardou, 2016-11-04:
Hey! I am trying to build cartographer using [these](https://google-cartographer-ros.readthedocs.io/en/latest/#building-installation) instructions but I get the following error. (my system is Ubuntu 14.04, gcc 4.8.4, ROS Indigo).

```
==> ninja -j1 in '/home/manos/Development/cartographer_ws/build_isolated/cartographer/install'
[2/296] Building CXX object cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o
FAILED: /usr/bin/clang++    -O3 -DNDEBUG -I. -I/home/manos/Development/cartographer_ws/src/cartographer -isystem /usr/src/gmock/gtest/include -I/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/{Boost_INCLUDE_DIRS}     -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o -MF "cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o.d" -o cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o -c /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc
In file included from /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc:17:
In file included from /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue.h:24:
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:60:7: warning: unknown attribute 'capability' ignored [-Wattributes]
class CAPABILITY("mutex") Mutex {
      ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:36:53: note: expanded from macro 'CAPABILITY'
#define CAPABILITY(x) THREAD_ANNOTATION_ATTRIBUTE__(capability(x))
                                                    ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:67:26: warning: unknown attribute 'acquire_capability' ignored [-Wattributes]
    Locker(Mutex* mutex) ACQUIRE(mutex) : mutex_(mutex), lock_(mutex->mutex_) {}
                         ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:48:33: note: expanded from macro 'ACQUIRE'
  THREAD_ANNOTATION_ATTRIBUTE__(acquire_capability(__VA_ARGS__))
                                ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:69:15: warning: unknown attribute 'release_capability' ignored [-Wattributes]
    ~Locker() RELEASE() {
              ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:51:33: note: expanded from macro 'RELEASE'
  THREAD_ANNOTATION_ATTRIBUTE__(release_capability(__VA_ARGS__))
                                ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:75:46: error: invalid use of 'this' outside of a non-static member function
    void Await(Predicate predicate) REQUIRES(this) {
                                             ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:45:53: note: expanded from macro 'REQUIRES'
  THREAD_ANNOTATION_ATTRIBUTE__(requires_capability(__VA_ARGS__))
                                                    ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:81:18: error: invalid use of 'this' outside of a non-static member function
        REQUIRES(this) {
                 ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:45:53: note: expanded from macro 'REQUIRES'
  THREAD_ANNOTATION_ATTRIBUTE__(requires_capability(__VA_ARGS__))
                                                    ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
In file included from /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc:17:
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue.h:110:33: warning: unknown attribute 'requires_capability' ignored [-Wattributes]
  bool QueueNotEmptyCondition() REQUIRES(mutex_) { return !deque_.empty(); }
                                ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:45:33: note: expanded from macro 'REQUIRES'
  THREAD_ANNOTATION_ATTRIBUTE__(requires_capability(__VA_ARGS__))
                                ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:57: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                                        ^
In file included from /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc:17:
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue.h:113:32: warning: unknown attribute 'requires_capability' ignored [-Wattributes]
  bool QueueNotFullCondition() REQUIRES(mutex_) {
                               ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:45:33: note: expanded from macro 'REQUIRES'
  THREAD_ANNOTATION_ATTRIBUTE__(requires_capability(__VA_ARGS__))
                                ^
[... 31 lines cut ...]
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                         ^
In file included from /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc:17:
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue.h:81:22: error: expected body of lambda expression
            [this]() REQUIRES(mutex_) { return QueueNotEmptyCondition(); },
                     ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:45:3: note: expanded from macro 'REQUIRES'
  THREAD_ANNOTATION_ATTRIBUTE__(requires_capability(__VA_ARGS__))
  ^
/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/mutex.h:31:42: note: expanded from macro 'THREAD_ANNOTATION_ATTRIBUTE__'
#define THREAD_ANNOTATION_ATTRIBUTE__(x) __attribute__((x))
                                         ^
5 warnings and 6 errors generated.
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/manos/Development/cartographer_ws/install_isolated/env.sh', 'ninja', '-j1']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/manos/Development/cartographer_ws/build_isolated/cartographer && /home/manos/Development/cartographer_ws/install_isolated/env.sh ninja -j1

Command failed, exiting.
```

Any idea what goes wrong?
Thanks!

SirVer, 2016-11-04:
Actually, your commandline shows that you are using clang, not gcc:

    FAILED: /usr/bin/clang++    -O3 -DNDEBUG -I. -I/home/manos/Development/cartographer_ws/src/cartographer -isystem /usr/src/gmock/gtest/include -I/home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/{Boost_INCLUDE_DIRS}     -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o -MF "cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o.d" -o cartographer/common/CMakeFiles/common_blocking_queue_test.dir/blocking_queue_test.cc.o -c /home/manos/Development/cartographer_ws/src/cartographer/cartographer/common/blocking_queue_test.cc

Unfortunately, Cartographer does not build with clang right now. And ROS is kinda gcc centric too. If you fix the build, please send a PR.

closed 2016-11-04 by SirVer

etsardou, 2016-11-04:
Correct! Thanks ;)