cartographer-project/cartographer_ros #166: have a error while i add IMU date for backpack_2d

carl1218, 2016-11-12:
hi,
 I have set the IMU drive in ros,and  changed the launch,when i runned the launch i got the following warn：
[ WARN] [1478952719.408758831]: W1112 20:11:59.000000  5699 tf_bridge.cc:51] "base_imu_link" passed to lookupTransform argument target_frame does not exist.

This is the lua：
```
include "map_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.,
    horizontal_laser_max_range = 20.,
    horizontal_laser_missing_echo_ray_length = 1.5,  constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "base_imu_link",
  published_frame = "laser_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.motion_filter.max_angle_radians = math.rad(0.1)
return options

```
This is the launch:

```
<launch>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename backpack_2d.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />
  </node>
<node pkg="razor_imu_9dof" type="imu_node.py" name="imu_node" output="screen">
   <param name="port" type="string" value="/dev/ttyUSB0"/>
   <param name="frame_id" type="string" value="base_imu_link" />
   <remap from="imu" to="/imu/data" />
</node>
</launch>
```

damonkohler, 2016-11-14:
My guess is that you've not changed your URDF to reflect the name of the IMU link "base_imu_link". This is more of a ROS issue than a Cartographer issue. I suggest taking followup questions to http://answers.ros.org/

closed 2016-11-14 by damonkohler

carl1218, 2016-11-17:
hi,
I added the IMU data in backpack_2d, but no matter how I placed the IMU in the direction（z or x or y-axis aligned wit word), I always get the Max inclination exceeded 20 warning:
```
[ WARN] [1479380979.407526014]: W1117 19:09:39.000000 78094 local_trajectory_builder.cc:241] Max inclination exceeded: 48.309 > 20
[ INFO] [1479380987.814185144]: I1117 19:09:47.000000 78094 collated_trajectory_builder.cc:79] imu rate: 48.76 Hz 2.05e-02 s +/- 2.00e-03 s (pulsed at 100.03% real time)

```

damonkohler, 2016-11-17:
Sounds like a separate issue. Please open a new issue and include your configuration, URDF, launch files, specs for the IMU, and a bag.

carl1218, 2016-11-17:
ok