cartographer-project/cartographer_ros #168: drift problems even no movement?

tom8382, 2016-11-14:
when i let my robot in static status, i found that the estimate pose drift at very very little step, however, this give me cumulative total error 20cm after about 30min in bnoth x and y axis. maybe i set motion filter too sensitive or sometiong else wrong, how to debug my error?? thank you

and also, i found that the estimate pose will Influence by anything moves around it. some other slam algorithm, may consider abnormal characteristic points out of match process. how about carto, it seems that no obvious function in carto for this?thank you

damonkohler, 2016-11-14:
We need to know more about your system to be able to answer this.

* Is this 2D or 3D?
* What's your configuration?
* What sort of environment?
* Can you provide a bag file?

Dynamic objects are handled by iterative "probability" updates to the map.

tom8382, 2016-11-15:
thank you for reply. drift problem not happened certainly,i will record a bag if it happened as possible.

damonkohler, 2016-11-15:
Ok, please open a new issue with as much information as possible if you need any help!

closed 2016-11-15 by damonkohler