cartographer-project/cartographer_ros #170: How to adjust the direction of the map for the horizontal and vertical ?

gogofight, 2016-11-17:
How to adjust the direction of the map for the horizontal and vertical from red to blue as follow ?

[image]

How to init map direction to any rotating degree or get it rightly ?

damonkohler, 2016-11-17:
You can apply that rotation in some image editing software if you like. The orientation of the map is determined by the first trajectory pose (i.e. the origin of the map). If you'd like the resulting map to be aligned, you can try starting in a position that is aligned with the walls.

closed 2016-11-17 by damonkohler

zhamengkeji, 2018-12-11:
@damonkohler I would like to ask how the generated map is implemented with the coordinates of the world map.

zhamengkeji, 2018-12-11:
@damonkohler How to achieve coordinates with world maps

zhamengkeji, 2018-12-11:
@damonkohler How to achieve alignment with world map coordinates

zhamengkeji, 2018-12-18:
@damonkohler 请问制图师发布的地图，怎么自动与世界地图坐标的对齐。

zhamengkeji, 2018-12-18:
@damonkohler Is this related to the initial coordinates?