cartographer-project/cartographer_ros #174: Running warn

carl1218, 2016-11-17:
Hi,
 I used the RAZOR 9DOF IMU.
 I added the IMU data in backpack_2d, but no matter how I placed the IMU in the direction（z or x or y-axis aligned wit word),and i  didn't move it ，I always get the Max inclination exceeded 20 warning:
```
[ WARN] [1479396992.674407112]: W1117 23:36:32.000000 80945 local_trajectory_builder.cc:241] Max inclination exceeded: 20.044 > 20
[ INFO] [1479397001.974362521]: I1117 23:36:41.000000 80945 submaps.cc:176] Added submap 2
[ INFO] [1479397001.984268399]: I1117 23:36:41.000000 80972 sparse_pose_graph.cc:258] We caught up. Hooray!
[ INFO] [1479397006.629708952]: I1117 23:36:46.000000 80945 collated_trajectory_builder.cc:79] imu rate: 48.76 Hz 2.05e-02 s +/- 1.94e-03 s (pulsed at 99.95% real time)
[ INFO] [1479397006.629845005]: I1117 23:36:46.000000 80945 collated_trajectory_builder.cc:79] scan rate: 9.96 Hz 1.00e-01 s +/- 4.70e-04 s (pulsed at 100.00% real time)
[ WARN] [1479397013.148004198]: W1117 23:36:53.000000 80945 local_trajectory_builder.cc:241] Max inclination exceeded: 46.4567 > 20
```
This is urdf:
```
<robot name="cartographer_backpack_2d">
<link name="base_link"/>
<link name="base_imu_link"/>
<link name="laser"/>
<joint name="joint_base_imu_link" type="fixed">
    <parent link="base_link" />
    <child link="base_imu_link" />
    <origin xyz="0 0 0" rpy="0 0 0" />
</joint>
<joint name="laser_joint" type="fixed">
    <parent link="base_link" />
    <child link="laser" />
    <origin xyz="0 0 0" rpy="0 0 0" />
</joint>
</robot>
```

This is lau:
```
include "map_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  sensor_bridge = {
    horizontal_laser_min_range = 0.,
    horizontal_laser_max_range = 6.,
    horizontal_laser_missing_echo_ray_length = 5,        
    constant_odometry_translational_variance = 0.,
    constant_odometry_rotational_variance = 0.,
  },
  map_frame = "map",
  tracking_frame = "base_imu_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = false,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2
```

This is launch:

  ```
<param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/backpack_2d.urdf" />
   <node name="robot_state_publisher" pkg="robot_state_publisher"
   type="robot_state_publisher" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename backpack_2d.lua"
      output="screen">
    <remap from="scan" to="/laser_scan" />
    <remap from="imu" to="/imu/data" />
  </node>
     <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>
```

damonkohler, 2016-11-18:
Can you provide a bag as well? Try enabling the IMU visualization in rviz. If your IMU is well aligned, the acceleration vector should be point straight up from the floor.

wlongxiang, 2016-11-18:
@carl1218 I also ran into this problem with my turtlebot 2 robot, and the robot model is inclined also in rviz.
I temporally solved this problem by disabling the imu in the configuration file, but I am not sure if it will affect the mapping

carl1218, 2016-11-18:
@damonkohler 
hi，
   The IMU is  positioned horizontally，and dind not move it,as shown in the picture.
This picture is IMU raw date:
[image]

This picture is the  Robot Initial stance:
[image]

IMU date look like OK，why always get the Max inclination exceeded 20 warning？

damonkohler, 2016-11-30:
Sorry for the delay.

The warning happens when the IMU estimate of orientation deviates more than 20 degrees from vertical. This can be caused by a few things:

- Your orientation is more than 20 degrees from vertical.
- Your IMU link in the URDF doesn't match reality and makes the system think your orientation is more than 20 degrees from vertical.
- Your IMU data isn't great and the orientation tracking is failing.

In the last case, you can try tuning TRAJECTORY_BUILDER_2D.imu_gravity_time_constant. Increasing it will trust the gyros more and the accelerometers less. Decreasing it will trust the accelerometers more and the gyros less. The perfect compromise will give you the best possible estimate of gravity for a given IMU.

closed 2016-11-30 by damonkohler

generalvisionyyt, 2017-01-21:
I meet this issue too on the kobuki mobile base's IMU data.I check the raw_data through rostopic echo and  find consistent exsiting errors of the x,y channels of the gyroscope.But on the other hand,there is no corresponding data on the x,y channel of the accelerometers for the exponential moving averaging algorithm to fix these errors.
So we solve this problem by just block these two channels of the gyroscopes since they are unused by the robot moving in the flat world.

carl1218, 2017-01-21:
@generalvisionyyt 
Did you solve this problem?
How did you block X and Y channels of the gyroscopes?
If we  block these two X,Y date of the gyroscopes , Whether to affect the use effect？