cartographer-project/cartographer_ros #176: build fails optimization_problem.cc

herrhannes, 2016-11-17:
Build fails in optimization_problem.cc with "undefined reference to ceres::..."
commit 0c8d6afc33531761e8dd69760907d730a8440ed3 still worked, did not try the commits in between. Also, there seems to be an issue with configuration_files_test.
I am using ROS Kinetic on Ubuntu 16.04LTS.

damonkohler, 2016-11-18:
Please provide the complete compilation output in a gist. Also, please provide the log from the configuration_files_test.

herrhannes, 2016-11-18:
This is the output from `catkin_make_isolated --install --use-ninja`: https://gist.github.com/herrhannes/2dda1b811e9672aa7af6f211c60c107e 
The error with configuration_files_test is in line 1399 of the gist.

tim-fan, 2016-11-21:
+1, I'm seeing the same build error. I'm also using ROS Kinetic and Ubuntu 16.04LTS. 

For me the build was passing as at commit https://github.com/googlecartographer/cartographer_ros/commit/e8b5dd6465dcea9bde47afd0496ca118bf063a10.

herrhannes, 2016-11-21:
Commenting out the configuration_files_test makes it compilable. But I am not sure if that's really a solution and not just a treatment for the symptom.

yoshito-okada, 2016-11-22:
> Commenting out the configuration_files_test makes it compilable.

This workaround worked for me too. I'm also using Ubuntu 16.04 + ROS kinetic.

BTW, my another computer having Ubuntu 14.04 + ROS jade could build the latest version as of today.

herrhannes, 2016-11-22:
I think, this could be part of both, reason and solution:
https://github.com/ros/catkin/pull/782

jeremymatthewwerner, 2016-11-29:
+1, I'm seeing the same build error. I'm also using ROS Kinetic and Ubuntu 16.04LTS.

redheli, 2016-11-30:
+1 same build error , ROS Kinetic and Ubuntu 16.04LTS

spend whole day try to fix the issue...

herrhannes, 2016-11-30:
I already opened an issue on github: https://github.com/googlecartographer/cartographer_ros/issues/176

 

Karlsruher Institut für Technologie (KIT)
Institut für Technik der Informationsverarbeitung (ITIV)

M. Sc. Hannes Stoll
Wissenschaftlicher Mitarbeiter

 
KIT – Die Forschungsuniversität in der Helmholtz-Gemeinschaft

redheli, 2016-12-01:
solution(not tested on Ubuntu 14.04):
add dependence **map_writer** to configuration_files_test

catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/CMakeLists.txt
```
google_test(configuration_files_test
  USES_CARTOGRAPHER
  SRCS
    configuration_files_test.cc
  DEPENDS
    map_writer
    node_options
)
```

jeremymatthewwerner, 2016-12-01:
That fixed it for me on Ubuntu 14.04 - thank you so much!
Any reason _not_ to merge that change in?

- Jeremy

jeremymatthewwerner, 2016-12-01:
Sorry I mean I'm on 16.04.

SirVer, 2016-12-05:
Sorry, I was on vacation. Will have a look this week.

ojura, 2016-12-07:
I can confirm the issue as well. Sounds to me that there are somewhere broken dependencies. I was able to get it to compile by taking a look at the linker invocation line (here's my build log: https://gist.github.com/ojura/fb9ab7c3a6b67dc99bfea3377ab40590) and adding additional linking calls to libceres.a after calls to libcartographer.a.

PR #222 (2016-12-09): Adds Kinetic to build matrix.
Fixes #176
closed 2016-12-09 by damonkohler