cartographer-project/cartographer_ros #179: location fail after a fast rotation．

tom8382, 2016-11-21:
[1121.bag.zip](https://github.com/googlecartographer/cartographer_ros/files/603073/1121.bag.zip)
[Uploading configurations.zip…]()

[image]
[image]
[image]

hi, i have met a problem that location fail after a fast rotation．here's my record bag with /scan /odom and configuration files. about 800 wall times it happened. i have imu and encoder on my robot, and fuse odom together with laser scan.please help me, thank you.

tom8382, 2016-11-21:
[configurations.zip](https://github.com/googlecartographer/cartographer_ros/files/603112/configurations.zip)
re-upload configurations.

tom8382, 2016-11-22:
[1122.bag.zip](https://github.com/googlecartographer/cartographer_ros/files/605203/1122.bag.zip)
i only test a fast rotation, and record a short bag, in this bag, i make carto only use odom to fusion with laser, but not rewrite odom. 1121.bag i rewrite odom to tf.

tom8382, 2016-11-22:
replay 1122.bag, i make  constant_odometry_translational_variance = 0.0. it got good result. but i other bag   such as 11.bag, it got bad result witch want variance = 1.5, how can i get it more robost??
[11.bag.zip](https://github.com/googlecartographer/cartographer_ros/files/605235/11.bag.zip)

damonkohler, 2016-11-30:
Sorry for the delay.

I had a look at your data and I have some questions:

- What sort of laser scanner is this? The pattern seems to measure multiple returns along the walls at times.
- Do you have IMU data? The scans appear distorted at times in ways that I assume are due to some instability in your robot base that could be measured by an IMU.

It actually looks like your laser scanner is slow enough and your base is fast enough that moving the base around causes significant distortion in the scans. In 3D, we support "unwarping" but that's not available in 2D right now. You could try unwarping the scans yourself using your odometry, but YMMV.

Significantly increasing the correlative scan matching search window improves things a bit. Your map doesn't get ripped apart at least:

```
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.3
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(45.)
```

tom8382, 2016-12-06:
- my laser is a single line 360 degree range,the driver is coded by myself. is the Laserscan message format wrong?

- i have imu data and encoder data in the odometry.

- i think you are right that my base odometry is about 200HZ, but laser only run at 6.3HZ. i am using base as the first odom source, and making laserscan  to correct the drift? maybe this is wrong? i should make cartographer as the only source of odom, and send imu and encoder data to cartographer for fusion with laserscan? 

can you explain more about unwarping the scans? i am not much understand. thank you!

damonkohler, 2016-12-06:
I'm not sure I follow exactly. Are you modifying the laser scan to correct for motion based on your odometry (i.e. unwarping it)? If so, you may want to try disabling that code. I'm not convinced it's working as intended. It's probably best if you feed all the raw data to Cartographer instead.

closed 2016-12-06 by damonkohler