cartographer-project/cartographer_ros #190: VLP-16 Modified Driver for Cartographer

xmfcx, 2016-11-25:
We have tried to get VLP-16 working with cartographer.

And we used this https://github.com/ros-drivers/velodyne driver to get data from vlp-16 and we are able to see point cloud on rviz in realtime.

But, It seems cartographer requires some modifications to this drivers in order to process it. And we are not sure what sort of things to change on it. And it also seems that **Google team** also used a VLP-16. 

So, is it possible for you to share the Velodyne driver you have used with us? If not, what sort of changes are we supposed to make in order to make it work?

Thanks.

damonkohler, 2016-11-28:
Why do you believe we require driver modifications? Although we've not tested with that driver specifically, I assume it's delivering PointCloud2 messages, which we ingest. You'll get better results if your driver delivers smaller point clouds from partial revolutions. We've had good luck with converting each individual UDP packet into separate clouds.

Kostech, 2016-12-16:
Hi there, I am having the same issue. Is this because that driver assigns the frame_id to the pointcloud as velodyne? 
Shall we manually change it to horizontal_vlp16_link?
What sort of modifications are required to make the cartographer subscribe to the data published from that driver?
Thank you!

damonkohler, 2016-12-19:
Please see the [ROS API documentation](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html) for configuring Cartographer to talk to your sensor.

SirVer, 2016-12-19:
@Kostech The secret sauce you probably want to learn about is topic remapping ([example in our docs](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/backpack_3d.launch#L29), but this is a ROS feature, not a Cartographer specific one).

yxting1994628, 2017-03-23:
Hellow!Have you solved this question?

alexwhittemore, 2017-09-27:
I'm on this same page at the moment - I've got a flood of pointclouds from my two VLP-16s and they're in separate frames, but published to the same topic which I think is an issue for Cartographer? It's unclear to me how to publish the two streams on separate topics.