cartographer-project/cartographer_ros #192: Building cartographer_rviz fails on NVIDIA Jetson TX1

5yler, 2016-11-25:
```
==> ninja -j4 -l4 in '/home/syler/google_ws/build_isolated/cartographer_rviz'
ninja: error: '/usr/lib/arm-linux-gnueabihf/libopencv_videostab.so.2.4.8', needed by '/home/syler/google_ws/devel_isolated/cartographer_rviz/lib/libcartographer_rviz_submaps_visualization.so', missing and no known rule to make it
<== Failed to process package 'cartographer_rviz': 
  Command '['/home/syler/google_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/syler/google_ws/build_isolated/cartographer_rviz && /home/syler/google_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.
```

I have OpenCV4Tegra (2.4.12) installed. I followed the steps in [ROS – Hacking for OpenCV on Nvidia Jetson TX1 & Jetson TK1](http://myzharbot.robot-home.it/blog/software/ros-nvidia-jetson-tx1-jetson-tk1-opencv-ultimate-guide/) to change the OpenCV search path for`image_geometry` and `cv_bridge` to the appropriate directories (`/usr/lib` instead of `/usr/lib/arm-linux-gnueabihf`) but got a different error when trying to build:

```
==> ninja -j4 -l4 in '/home/syler/google_ws/build_isolated/cartographer_rviz'
[3/16] Automoc for target drawable_submap
Generating moc_drawable_submap.cpp
[6/16] Automoc for target submaps_display
Generating moc_submaps_display.cpp
[10/16] Building CXX object cartographer_rviz/CMakeFiles/submaps_display.dir/submaps_display.cc.o
FAILED: /usr/bin/c++   -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_rviz\" -O3 -DNDEBUG -fPIC -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtGui -isystem /usr/include/qt4/QtCore -isystem /home/syler/google_ws/install_isolated/include -isystem /opt/ros/indigo/include -isystem /usr/include/eigen3 -isystem /usr/include/OGRE -isystem /usr/include/opencv -isystem /usr/include/lua5.2 -I. -I/home/syler/google_ws/src/cartographer_ros/cartographer_rviz -I/home/syler/google_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz     -pthread -std=c++11  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=uninitialized -O3 -DNDEBUG -MMD -MT cartographer_rviz/CMakeFiles/submaps_display.dir/submaps_display.cc.o -MF "cartographer_rviz/CMakeFiles/submaps_display.dir/submaps_display.cc.o.d" -o cartographer_rviz/CMakeFiles/submaps_display.dir/submaps_display.cc.o -c /home/syler/google_ws/src/cartographer_ros/cartographer_rviz/cartographer_rviz/submaps_display.cc
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
[10/16] Building CXX object cartographer_rviz/CMakeFiles/submaps_display.dir/submaps_display_automoc.cpp.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_rviz': 
  Command '['/home/syler/google_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/syler/google_ws/build_isolated/cartographer_rviz && /home/syler/google_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.
```

SirVer, 2016-12-05:
> c++: internal compiler error: Killed (program cc1plus)

You have a crash in your compiler. Sorry, that is out of scope for Cartographer. Maybe try a different compiler version? Or you are running out of memory in the compiler, could also be.

closed 2016-12-05 by SirVer

carl1218, 2016-12-11:
@5yler 
hi , i also got this error，Have you solved the error？

rohbotics, 2016-12-11:
Looks like you are running out of ram.
Check `top` (or `htop`) to find out how much memory you are using when the compiler errors. 
You might want to try adding a swap file/disk, I use a usb pendrive with a swap partition.

It is not the fastest, but it does let it build on low(er) memory systems.

5yler, 2016-12-12:
@carl1218 @SirVer I went through the [ROS – Hacking for OpenCV on Nvidia Jetson TX1 & Jetson TK1](http://myzharbot.robot-home.it/blog/software/ros-nvidia-jetson-tx1-jetson-tk1-opencv-ultimate-guide/) guide again and double-checked that I made all the necessary modifications:

``` 
sudo cp /opt/ros/indigo/lib/pkgconfig/cv_bridge.pc /opt/ros/indigo/lib/pkgconfig/cv_bridge.pc-bak
sudo cp /opt/ros/indigo/lib/pkgconfig/image_geometry.pc /opt/ros/indigo/lib/pkgconfig/image_geometry.pc-bak
sudo cp /opt/ros/indigo/share/cv_bridge/cmake/cv_bridgeConfig.cmake /opt/ros/indigo/share/cv_bridge/cmake/cv_bridgeConfig.cmake-bak
sudo cp /opt/ros/indigo/share/image_geometry/cmake/image_geometryConfig.cmake /opt/ros/indigo/share/image_geometry/cmake/image_geometryConfig.cmake-bak
sudo gedit /opt/ros/indigo/lib/pkgconfig/cv_bridge.pc &
sudo gedit /opt/ros/indigo/lib/pkgconfig/image_geometry.pc &
sudo gedit /opt/ros/indigo/share/cv_bridge/cmake/cv_bridgeConfig.cmake &
sudo gedit /opt/ros/indigo/share/image_geometry/cmake/image_geometryConfig.cmake &
```
> **Modifications for each file**
>1. remove each instance “/usr/lib/arm-linux-gnueabihf/libopencv_ocl.so.2.4.8;“
>2. replace each instance of “/usr/lib/arm-linux-gnueabihf/” with “/usr/lib“
>3. replace each instance of “2.4.8” with “2.4.12” (or the current version of OpenCV in opencv4tegra package)

I think I must have missed something the previous time I tried, as the build just completed successfully!

carl1218, 2016-12-12:
@5yler . it is very strange, In my system there is no such file about cv_bridge.pc.  You are using the Ubuntu 64 bit system or 32 bit？

5yler, 2016-12-12:
@carl1218 The NVIDIA Jetson is ARM-based, so neither 32 not 64 bit. See [32-bit vs 64-bit vs ARM in regards to programs and OSes](http://unix.stackexchange.com/a/125314) on unix.stackexchange.com. You may want to open a separate issue, since it appears my problem was specific to the OpenCV library linking with NVIDIA's non-standard OpenCV4Tegra distribution that led to problems.

ZhiyuLiu1993, 2018-05-22:
@SirVer Could you tell me why cartographer does not support arm32?  
My platform is  Ubuntu 14.04.3 LTS (GNU/Linux 3.4.39 armv7l),  CPU is ARMv7 Processor rev 5 (v7l). I want to try run cartographer on my platform.  I want to know what exactly wrong and  what should I do to achieve my purpose?  Thanks!

SirVer, 2018-05-23:
@ZhiyuLiu1993 Cartographer is only tested on 64bit hardware (see the CI runs), so it might work on 32bit arm, but there is neither guarantee nor verification that it does. It might.

For example, we know that Eigen requires alignment macros for 32bit systems which last time I checked Cartographer had not in place. So it might not work.

ZhiyuLiu1993, 2018-05-23:
@SirVer Thank you for your answer. I will try it on the 64-bit arm in the future.