cartographer-project/cartographer_ros #193: MultiSense SL

athtest800, 2016-11-27:
Hello!
Does cartographer work with MultiSense SL Head?
[http://carnegierobotics.com/multisense-sl/](http://carnegierobotics.com/multisense-sl/)
Has anybody an example?
Thank you.

damonkohler, 2016-11-28:
@skohlbr is doing something similar. As of #185 we support axially rotated scanners for 3D SLAM.

closed 2016-11-28 by damonkohler

skohlbr, 2016-11-28:
I actually even should have some Multisense SL bag files of Atlas walking around. I'm a bit busy in the next few days though, so will only be able to look for one later.

damonkohler, 2016-11-30:
@skohlbr It would be great to have a configuration checked in for the Multisense. We're happy to host the bag :)

athtest800, 2017-01-01:
@skohlbr Do we have some news on Multisense?
Thanks!