cartographer-project/cartographer_ros #196: Sensor Bridge Error

athtest800, 2016-11-28:
I get this error:

> [ INFO] [1480358997.125607301]: I1128 20:49:57.000000 22055 submaps.cc:329] Added submap 1
[ INFO] [1480358997.125663927]: I1128 20:49:57.000000 22055 map_builder_bridge.cc:37] Added trajectory with ID '0'.
F1128 20:49:57.308058 22055 sensor_bridge.cc:61] Check failed: msg->linear_acceleration_covariance[0] != -1 (-1 vs. -1) 
[FATAL] [1480358997.308333001]: F1128 20:49:57.000000 22055 sensor_bridge.cc:61] Check failed: msg->linear_acceleration_covariance[0] != -1 (-1 vs. -1) 
*** Check failure stack trace: ***
    @     0x7fb3af9b25cd  google::LogMessage::Fail()
    @     0x7fb3af9b4433  google::LogMessage::SendToLog()
    @     0x7fb3af9b215b  google::LogMessage::Flush()
    @     0x7fb3af9b4e1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x53fb14  cartographer_ros::SensorBridge::HandleImuMessage()
    @           0x51ab74  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros4Node10InitializeEvEUlRKNS_10shared_ptrIKN11sensor_msgs4Imu_ISaIvEEEEEE2_vSD_E6invokeERNS1_15function_bufferESD_
    @           0x525ade  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x52e639  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7fb3b0ca35cd  ros::SubscriptionQueue::call()
    @     0x7fb3b0c4dcf0  ros::CallbackQueue::callOneCB()
    @     0x7fb3b0c4f0f3  ros::CallbackQueue::callAvailable()
    @     0x7fb3b0ca7691  ros::SingleThreadedSpinner::spin()
    @     0x7fb3b0c8c72b  ros::spin()
    @           0x4f2249  cartographer_ros::(anonymous namespace)::Run()
    @           0x4f0454  main
    @     0x7fb3ac83e830  __libc_start_main
    @           0x4f1ec9  _start
    @              (nil)  (unknown)

athtest800, 2016-11-29:
FIXED: IMU was not publishing data

closed 2016-11-29 by athtest800

christiann017, 2023-08-26:
Hallo, i want to know.
i found a problem like this. how to handle it?