cartographer-project/cartographer_ros #202: cartographer_node is not publishing 'map' tf frame. Fixed frame map does not exist.

sockhands, 2016-11-30:
I am trying to test cartographer using my own bag file. 
I have filtered the bag file so that it will only play back the topics /imu, /scan and /odom.

Running tf view_frames shows the following hierarchy (top to bottom):
base_link -> laser -> imu_link
...which corresponds with how i've set the urdf file.

As I understand it, base_link should then be the child of odom (with provide_odom_frame set to true), which should be the child of map (map being top of the tree), but both of these are missing from the tf tree.

The /map topic exists, but rostopic echo /map comes up quiet. 
Are there any general reasons why a map frame might not get published? I thought at the very least, the tf frame should be there.

 Rqt_graph comes up exactly as it does in backpack_2d demo, except cartographer is subscribed to /scan, /imu and /odom from the bag.

```
include` "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry = false,
  use_laser_scan = false,
  use_multi_echo_laser_scan = true,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true

return options``

damonkohler, 2016-12-01:
It sounds like SLAM isn't actually running. The cartographer_node will publish the /map and /odom frames. There is no topic /map, all frames are published to the /tf topic. Have a look at the console output and see if cartographer_node is giving you some clue as to what the issue is (e.g. not publishing sensor data).

Overall, this sounds like more of a ROS issue than a Cartographer issue. I suggest trying http://answers.ros.org/. I'll close this for now. If you find you're having an issue with Cartographer, please feel free to reopen or open a new issue.

sockhands, 2016-12-01:
Thanks for the response. Turns out my imu is actually publishing the base_link frame. I was arbitrarily putting an imu_link in my urdf, but I've since removed it, and my tf tree without cartographer running is simply base_link -> laser. 

SLAM works now, and I get the map and odom transformations from cartographer, but only when I set use_imu_data to false in the lua config. With use_imu_data set to true, cartographer subscribes to /imu, and imu data is being published, but as before, cartographer stops providing map and odom transforms. 

In the console, Cartographer_node simply says that all sensor data is available, then produces no further logs, good or bad. Can cartographer silently reject the data from the imu?

damonkohler, 2016-12-02:
Please share the console output from Cartographer and from roswtf (run it while Cartographer is stalled) in a gist.

sockhands, 2016-12-03:
Having investigated, it appears the problem is on my end. I don't have the computer available to me currently, so can't post logs, but looking at the imu data in rviz revealed totally erratic behaviour. IMU was likely not configured properly. Will post logs as as I'm able.

damonkohler, 2016-12-07:
I'll close this again for now. Please let us know if you need more assistance.

closed 2016-12-07 by damonkohler

blairlpp, 2016-12-20:
Oh my god! I tried cartographer on my own robot and confronted the same problem. How do you solve it?

SirVer, 2016-12-20:
@blairlpp Please file a new issue and provide the asked for information in the bug template.