cartographer-project/cartographer_ros #207: map rotates continuously

athtest800, 2016-12-01:
Hello again.
I try to use hokuyo laser.
In 2D all OK but in 3D the map rotates continuously clockwise, in steps of 10 degrees, without moving the laser.
Is it common problem? The IMU gives stable data.
Thank you

damonkohler, 2016-12-02:
Are you trying to use a 2D LIDAR for 3D? That won't work unless it's a spinning LIDAR (e.g. Multisense SL). See the taurob_tracker.lua example for that sort of configuration.

athtest800, 2016-12-02:
Yes. I try to use 2D LIDAR for 3D.
I used the taurob_tracker.lua by removing the odom as long as i have no odometry.
Here is the bag 

> https://github.com/athtest800/cartographer/blob/master/cartographer_ros/launch/record.bag?raw=true.

 Can you please check it? The map still rotates.

damonkohler, 2016-12-02:
Is your 2D LIDAR axially rotating? Also, I'm skeptical about getting it to work without odometry.

athtest800, 2016-12-02:
Yes, it rotates axially like the [.](http://carnegierobotics.com/multisense-sl/)
I can not have odometry as it is handheld. Do you recommend to use visual odometry?
Have you checked the bag?

damonkohler, 2016-12-02:
I'll have to give it a try next week. Maybe it will work? Will be interesting to find out :)

damonkohler, 2016-12-06:
I had a look at the bag and determined that your IMU gyros are either significantly biased or otherwise not very good. You can improve the results by turning down the reliance on the gyros for scan matching:

```
TRAJECTORY_BUILDER_3D.ceres_scan_matcher.rotation_weight = 1
```

You may have to do the same thing for the global problem (see `SPARSE_POSE_GRAPH.optimization_problem.rotation_weight`) but the bag was too short to test this.

closed 2016-12-06 by damonkohler