cartographer-project/cartographer_ros #210: bag display error after solving #176 issue

AhmedElsayedHamouda, 2016-12-04:
hi there, i'm new in ros field, im try to use cartographer package.
the steps use wstool, after doing it, problem solved in issue #176 happened, and solved using the solution in the the issue, after downloading the bag, another error happened after roslaunch command "roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag":
 

[FATAL] [1480843719.721312649]: Error opening file: /home/ahmed/cartographer_paper_deutsches_museum.bag
[playbag-4] process has died [pid 9445, exit code 1, cmd /opt/ros/kinetic/lib/rosbag/play --clock /home/ahmed/cartographer_paper_deutsches_museum.bag __name:=playbag __log:=/home/ahmed/.ros/log/043b44c6-ba04-11e6-87ec-18f46a059ed7/playbag-4.log].
log file: /home/ahmed/.ros/log/043b44c6-ba04-11e6-87ec-18f46a059ed7/playbag-4*.log
[ INFO] [1480843720.389725379]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1480843720.462560762]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1480843720.462715754]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1480843720.466753572]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1480843720.466853804]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1480843720.490070974]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1480843720.490227783]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1480843720.506834107]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1480843720.506955723]: I1204 11:28:40.000000  9443 configuration_file_resolver.cc:40] Found '/home/ahmed/catkin_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1480843720.642808468]: I1204 11:28:40.000000  9443 submaps.cc:176] Added submap 1
[ INFO] [1480843720.642953939]: I1204 11:28:40.000000  9443 map_builder_bridge.cc:37] Added trajectory with ID '0'.

any ideas, thanks.

damonkohler, 2016-12-05:
The logs don't match your description. "Error opening file: /home/ahmed/cartographer_paper_deutsches_museum.bag" indicates that you didn't specify your ~/Downloads" location. It looks to me like you've simply specified the incorrect location of the downloaded bag file on the command line.

closed 2016-12-05 by damonkohler

AhmedElsayedHamouda, 2016-12-06:
thanks for replying, the path was wrong