cartographer-project/cartographer_ros #212: Several laser scanners

mgvargas, 2016-12-05:
Hi! 
I'm trying to use google cartographer with a robot that has two 2D Hokuyo laser scanners to obtain a 360° view around the robot. Currently, cartographer supports  only one LaserScan topic. I was trying to combine both sensor readings, but the results don't always correctly map the real environment.

Is it possible to implement a new functionality that supports more than one laser scanner?

 Thanks

damonkohler, 2016-12-05:
This is definitely possible.

Implementation in 3D for axially rotated scanners should be rather straight forward. Simply follow the example set by PointCloud2 topics.

Implementation in 2D will require more work. Look to the 3D implementation for an [example of how scans are accumulated](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_3d/kalman_local_trajectory_builder.cc#L85). Something similar would need to be added to the [2D trajectory builder](https://github.com/googlecartographer/cartographer/blob/master/cartographer/mapping_2d/local_trajectory_builder.cc#L152).

samrobotics, 2017-01-09:
There is an implementation of a node that merges multiple scan topic into a single scan available at https://github.com/iralabdisco/ira_laser_tools

I have been using this successfully with cartographer in a similar setup to the one you describe (two Sick TiM 561) for several months both in simulation and on real robots without any major issues :-)

closed 2017-07-21 by wohe

Samer-Docs, 2021-06-22:
> This is definitely possible.
>
> Implementation in 3D for axially rotated scanners should be rather straight forward. Simply follow the example set by PointCloud2 topics.
>
> Implementation in 2D will [...]

Hi Mr. Damonkohler,

Thanks for your comment. I tried to follow, but the links don't work. 

I have a similar issue. I have a combination of three Hokuyo scanners (one horizontal, two slanted) mounted on a backpack platform. I am trying to run Cartographer 3D on this configuration.  Is it possible?

kind regards

PR #435 (2017-07-21): Support multiple LaserScan and MultiEchoLaserScan topics.
This fixes #212.
commit e5c2881b (2017-07-21): Support multiple LaserScan and MultiEchoLaserScan topics. (#435)

This fixes #212.