cartographer-project/cartographer_ros #224: 0 computations resulted in 0 additional constraints

redheli, 2016-12-10:
Hi,

  I tried cartographer with several dataset (1. handhold velodyne puck with imu, 2. on vehicle velodyne puck and imu), 
   the scan matching result in rviz is amazing , and the trajectory is almost follow the ground truth, but not converge when revisit same place.

   I also notice on the terminal , it always print out below info, seems no close loop found and optimize failed?

   Thank you.

> 
> [ INFO] [1481267748.641034749]: I1209 15:15:48.000000 26526 sparse_pose_graph.cc:270] We caught up. Hooray!
> [ INFO] [1481267748.982431370]: I1209 15:15:48.000000 26468 submaps.cc:329] Added submap 2
> [ INFO] [1481267748.987050203]: I1209 15:15:48.000000 26523 constraint_builder.cc:304] 0 computations resulted in 0 additional constraints.
> [ INFO] [1481267748.987104260]: I1209 15:15:48.000000 26523 constraint_builder.cc:306] Score histogram:
> Count: 0
> [ INFO] [1481267749.009439539]: I1209 15:15:49.000000 26523 sparse_pose_graph.cc:270] We caught up. Hooray!
> [ INFO] [1481267749.409317543]: I1209 15:15:49.000000 26527 constraint_builder.cc:304] 0 computations resulted in 0 additional constraints.
> [ INFO] [1481267749.409369854]: I1209 15:15:49.000000 26527 constraint_builder.cc:306] Score histogram:
> Count: 0

my dataset one:
1) [handhold bag](https://www.dropbox.com/s/smugu647f1u6pv4/car_velodyne_imu.bag?dl=0) (in this bag file the imu topic frame start with "/", I use [imu_remove_splash.py](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/scripts/imu_remove_splash.py) to remove "/" and republish the msg with topic "imu_no_splash"
2) [launch file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/launch/demo_mybag.launch)
3) [urdf file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/urdf/my_velodyne_imu_bag.urdf)
4) [lua file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/configuration_files/my_velodyne_imu_bag.lua)

dataset two:
1) [car bag](https://www.dropbox.com/s/ux96p7bjmgab7un/gates_oscillating_motion.bag?dl=0)
2) [launch file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/launch/car_velodyne_imu.launch)
3) [urdf file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/urdf/car_velodyne_imu.urdf)
4) [lua file](https://github.com/redheli/cartographer_ros/blob/master/cartographer_ros/configuration_files/car_velodyne_imu.lua)

SirVer, 2016-12-15:
​You had me quite confused here, because you swapped the links of your bag files. ​Once I figured this out, it was smooth sailing getting your stuff up and running. Kudos for putting your changes in a git repo, it made reproducing much easier for us! I wish everybody asking for help did that.

I only looked at the car bag file, I think you can probably tune the system yourself for the other bag with my changes.

I pushed my changes to my repo, in-lining comments explaining why I did the changes: https://github.com/SirVer/cartographer_ros/commit/72c45195d55b0a4176afef5c7db1d44c0693527f

With that I got reasonable loop closing - I think it could be improved with further tuning, but it seems a good starting point:

[image]

closed 2016-12-15 by SirVer

redheli, 2016-12-16:
@SirVer Thank you so much! Reading you comments now.. It is precious.