cartographer-project/cartographer_ros #227: How to disable imu in 3D map building?

jvlevia, 2016-12-13:


wohe, 2016-12-13:
This is currently not supported. That said, depending on what you are trying to achieve you might be able to use 2D SLAM without IMU, or provide fake IMU data for 3D SLAM. What kind of sensor configuration are you using? In what kind of environment do you want to use it?

jvlevia, 2016-12-14:
@wohe Hi, thanks for your reply. I am using velodyne 32E, basically we want to drive the car in the campus to build the map. May I ask how to generate fake IMU data ?

damonkohler, 2016-12-14:
If you have a 32 beam Velodyne, surely you can also add on an IMU? Generating fake data will not generate very good results. As @wohe said, it's not supported.

jvlevia, 2016-12-16:
Hi, I have recorded the campus with imu data, this is the result, why is it not correct ?
[image]

SirVer, 2016-12-16:
This is impossible to say from this picture alone. Maybe your IMU frame is wrong in your URDF? 

Please push a repository with the configuration files you are using to run Cartographer and rviz and provide us with a bag file of sample data so we can investigate what is wrong.

jvlevia, 2016-12-19:
The lidar & imu are mounted right on top of the car, not tilted. 
I try to compare the imu data between the example bagfile & my own bagfile, I found out that there is some differences, is it possible that these are the causes ?

Example bagfile's imu data :
[image]

My own bagfile's imu data :
[image]

damonkohler, 2016-12-19:
Your IMU appears to be removing gravity from your linear acceleration. Cartographer requires that it be left in.

SirVer, 2016-12-19:
Damon saw something obviously wrong in the data, but all in all this error analysis is guesswork until we see what you actually did for configuration:

Please push a repository with the configuration files you are using to run Cartographer and rviz and provide us with a bag file of sample data so we can investigate what is wrong.
￼  ￼  ￼

jvlevia, 2016-12-20:
@damonkohler Is there any way that I can add the gravity to my linear acceleration data?

jvlevia, 2016-12-20:
@SirVer  Hi I am new to github, very sorry that I don't know how to push a repository. 

**# Here is my bagfile :** 
https://drive.google.com/open?id=0B8yQMm4hq9Z1bzhWMmVPNjgzb3M

# Launch file : 

`<launch>
  <param name="/use_sim_time" value="true" />

  
  <arg name="base_frame" default="/base_link"/>
  <arg name="topic_name" default="vehicle_model"/>
  <arg name="offset_x" default="0.0"/> 
  <arg name="offset_y" default="0.0"/>
  <arg name="offset_z" default="0.0"/>
  <arg name="offset_roll" default="0.0"/> 
  <arg name="offset_pitch" default="0.0"/> 
  <arg name="offset_yaw" default="0.0"/> 
  <arg name="model_path" default="$(find cartographer_ros)/urdf/coms_aw.urdf" />  
  <arg name="gui" default="False" />
  <param name="robot_description" textfile="$(arg model_path)" />
  <param name="use_gui" value="$(arg gui)"/>
  <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher" />
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="state_publisher" /> 

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename toyota_3d.lua"
      output="screen">
    <remap from="points2" to="points_raw" />
    <remap from="imu" to="imu/data"/>
  </node>

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_3d.rviz" />
  <node name="playbag" pkg="rosbag" type="play"
      args="--clock $(arg bag_filename)" />
</launch>`

# Lua file :
`include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry = false,
  use_laser_scan = false,
  use_multi_echo_laser_scan = false,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

TRAJECTORY_BUILDER_3D.scans_per_accumulation = 160

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
MAP_BUILDER.sparse_pose_graph.optimization_problem.huber_scale = 5e2
MAP_BUILDER.sparse_pose_graph.optimize_every_n_scans = 320
MAP_BUILDER.sparse_pose_graph.constraint_builder.sampling_ratio = 0.03
MAP_BUILDER.sparse_pose_graph.optimization_problem.ceres_solver_options.max_num_iterations = 10
-- Reuse the coarser 3D voxel filter to speed up the computation of loop closure
-- constraints.
MAP_BUILDER.sparse_pose_graph.constraint_builder.adaptive_voxel_filter = TRAJECTORY_BUILDER_3D.high_resolution_adaptive_voxel_filter
MAP_BUILDER.sparse_pose_graph.constraint_builder.min_score = 0.62

return options`

# URDF file :

`<?xml version="1.0"?>

<robot name="car">
  <link name="base_link">
    <visual name="base_visual">
      <origin xyz="-0.6 0 0" rpy="0 0 -1.571" />
      
      <material name="white">
         <color rgba="1 1 1 1"/>
      </material>
      <geometry>
        <box size="0.7200 2.2025 0.1015"/>
        <mesh filename="package://model_publisher/../../../.config/model/mesh/coms.dae"/>
      </geometry>
    </visual>
  </link>
  
  <link name="velodyne">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.07" radius="0.05" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link_novatel">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

  <joint name="velodyne_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="velodyne" />
    <origin xyz="-1.04 0. 1.71" rpy="0.0 0.0 -0.04363" />
  </joint>

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="base_link_novatel" />
    <origin xyz="0 0 0" rpy="0 0 0" />  
  </joint>
 
</robot>`

SirVer, 2016-12-20:
Your copy and pasted code misses characters like `<>`. I suggest you learn GitHub - which is very well documented - or upload all data and files somewhere in its entirety.

SirVer, 2016-12-20:
I am closing this for now since we figured out that your IMU is lying about the physical accelerations it measures. This is therefore not a Cartographer issue.

closed 2016-12-20 by SirVer

ludennis, 2018-09-01:
Hi all,

Is there a way to turn off the IMU correction done by `optimization_problem_3d.cc`? I have used the IMU correction from Cartographer and modify my `.urdf` file accordingly, but now with a corrected IMU Cartographer still tries to correct it. So I am wondering if there's a way to turn it off. 

I have posted a [video](https://drive.google.com/file/d/1HekIFolrUCxRvgCT5dUL3HMRAJtdyc5X/view) that shows the correction in  [Cartographer issue #440](https://github.com/googlecartographer/cartographer/issues/440#issuecomment-417782519)