cartographer-project/cartographer_ros #248: cartographer not publishing /map frame

kentsommer, 2017-01-16:
# Overview
Been running into a bit of a wall with getting cartographer up and running on the kitti dataset and was hoping I'm just missing something obvious. 

1. You can find all the required launch and configuration files on my [fork](https://github.com/kentsommer/cartographer_ros) under the "kitti" branch.

2. rosbag file can be found [here](https://drive.google.com/file/d/0B5DVPd_zGgc8NElwN0JrczAxYTQ/view?usp=sharing) (~2GB)

It seems that the map frame is not getting published in /tf which would make me think that SLAM isn't running for some reason. However, after futzing around with this for the past day and not making much progress I thought I would ask for some assistance as maybe I'm overlooking something simple. 

# How I am launching

```
roslaunch cartographer_ros demo_kitti_3d.launch bag_filename:=${HOME}/Downloads/kitti_2011_09_26_drive_0009_sync.bag
```

# Possibly helpful debug output

**output of roswtf with cartographer "running"**

```
kent@ksbox ~/Documents/workspaces/cartographer $ roswtf
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

No errors or warnings
```

**Beginning output after launching**

```
kent@ksbox ~/Documents/workspaces/cartographer/src/cartographer_ros/cartographer_ros/launch $ roslaunch cartographer_ros demo_kitti_3d.launch bag_filename:=${HOME}/Downloads/kitti_2011_09_26_drive_0009_sync.bag
... logging to /home/kent/.ros/log/74123c58-dbbc-11e6-9e74-94de8023c908/roslaunch-ksbox-10231.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ksbox:43144/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    playbag (rosbag/play)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [10335]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 74123c58-dbbc-11e6-9e74-94de8023c908
process[rosout-1]: started with pid [10424]
started core service [/rosout]
process[cartographer_node-2]: started with pid [10438]
process[rviz-3]: started with pid [10447]
[ INFO] [1484551313.331674886]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer_ros/configuration_files/kitti_3d.lua' for 'kitti_3d.lua'.
[ INFO] [1484551313.331943340]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1484551313.331981469]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1484551313.332045720]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1484551313.332086826]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1484551313.332210513]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1484551313.332248303]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1484551313.332366140]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1484551313.332399676]: I0116 16:21:53.000000 10438 configuration_file_resolver.cc:40] Found '/home/kent/Documents/workspaces/cartographer/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
process[playbag-4]: started with pid [10469]
[ INFO] [1484551313.365571278]: I0116 16:21:53.000000 10438 submaps.cc:329] Added submap 1
[ INFO] [1484551313.365719450]: I0116 16:21:53.000000 10438 map_builder_bridge.cc:37] Added trajectory with ID '0'.
0x17735a0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12c6110) ): Attempt to set a screen on a child window.
0x177c470 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12c6110) ): Attempt to set a screen on a child window.
0x13f6f70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12c6110) ): Attempt to set a screen on a child window.
0x1783d80 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12c6110) ): Attempt to set a screen on a child window.
0x177c470 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x1783d80 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x13f6f70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x1ddb280 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x1f101f0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x1f3e5e0 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
0x1f47f70 void QWindowPrivate::setTopLevelScreen(QScreen*, bool) ( QScreen(0x12bad00) ): Attempt to set a screen on a child window.
[ WARN] [1484551326.866681962, 1317010105.207919753]: W0116 16:22:06.000000 10438 tf_bridge.cc:51] "imu" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1484551327.072596807, 1317010105.414539705]: W0116 16:22:07.000000 10438 tf_bridge.cc:51] "imu" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1484551327.277159855, 1317010105.618405315]: W0116 16:22:07.000000 10438 tf_bridge.cc:51] "imu" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1484551327.478946200, 1317010105.820345830]: W0116 16:22:07.000000 10438 tf_bridge.cc:51] "imu" passed to lookupTransform argument target_frame does not exist. 
```

kentsommer, 2017-01-17:
Update:

I believe the issue was possibly due to the 'world' tf frame and after filtering it out the 'map' and 'odom' frames are published. 

All changes/fixes are pushed to my fork under the kitti branch if someone ends up running into something similar.

damonkohler, 2017-01-17:
@kentsommer Glad to hear you got it working for you. We're very interested in your results and would love to provide configurations for the Kitti dataset as an example. Is it working well for you? Could you craft a PR?

kentsommer, 2017-01-17:
@damonkohler Does not seem to be working well, and I was toying with re-opening this to have someone else look at what might be going on. 

Working theory right now is that cartographer has a really difficult time initializing with any of the kitti runs. I'll try recording a short video and post it here so you can take a look.

SirVer, 2017-01-17:
Reopening, since investigating Kitti is something we are interested in. @kentsommer anything you have would be much appreciated as a jump-off point.

kentsommer, 2017-01-17:
@SirVer all launch + configuration files can be found on the kitti branch of my [fork](https://github.com/kentsommer/cartographer_ros)

kentsommer, 2017-01-18:
@SirVer @damonkohler 

Here is the video I promised. Sorry about the flickering of the wallpaper, funky recording artifact. 

Things to note:
* The map and odom frames are not published for a fairly long amount of time (hence the initialization issue theory)
* Scan matched points is overly dense
* Seems like position estimation "catches" up at times of slower movement or stops and you can see the base_link "jump" a little bit

SirVer, 2017-01-19:
I will only have time next week to look into this. The video will be helpful. Did you run roswft to figure out if all your frames are connected?

kentsommer, 2017-01-22:
@SirVer 

Yeah, and confirmed in Rviz as well. Here is the output of roswtf:

```
kent@ksbox ~/Documents/workspaces/cartographer $ roswtf
Loaded plugin tf.tfwtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

No errors or warnings
```

12tbuchman, 2017-01-24:
im also running into a similar problem as what you had in the beginning. My map frame isnt being published to tf. I am using a hokuyo laser scanner. I ran my setup against a bag file and it worked fine, but when I use the laser scanner, Cartographer isnt publishing any scan matching or map.

SirVer, 2017-01-26:
@12tbuchman Your problem is probably different (see below). Please open a new bug report and provide configuration and bags.

@kentsommer Sorry for taking so long to get back to you. And also for asking you for information you have already provided - I lost the overview of open issues.

I made your example work (https://github.com/googlecartographer/cartographer_ros/commit/1af94cc57f8e7b4827acafe956f65288b848f657), generating this map:

[image] 

This looks reasonable, but not great. Here are some assorted thoughts:

- The converter you used to generate the Velodyne point clouds is outputting full revolutions into one PointCloud2. This is sub-optimal. For Cartographer, you'd want it to output much smaller subdivisions of a revolution and let Catographer do the accumulation (`scans_per_accumulation` setting), because Cartographer can use its motion model to "unwarp" the accumulated point cloud, i.e. take the 'running shutter' effect into account. 
- In your config you had the setting on 160 still, which we use for our demo bags where each UDP packet from the VLP16 is translated into one PointCloud2. This means you waited for 160 revolutions before attempting a single scan match.
- I think the non-unwarping is one of the main reasons the map does not look too great: at a movement speed of 50 km/h and a laser rotation of 10 Hz, you'd have ~1.4m error through the movement.
- It is fairly ambitious to try slamming at 5cm for this laser, its noise and the mean distances. I would suggest upping to at least 10 or 15cm.
- The bag was too short to really asses parameters for loop closing. The local SLAM looked alright and before the unwarping of the revolutions is not done, there is not much point in trying to tweak parameters for local SLAM.
- Our RViz visualization plug in does not deal well with open skies: We try to approximate the XRay look by finding a "ceiling" and a "floor" and doing the XRay math on that. Under open skies we do not find a ceiling, so the map only has the walls, but no free space. This is purely an optical issue in RViz and does not affect the SLAM though.

Closing this, as SLAM is basically working. If you require more help, just let me know. 

Fun fact: I was at the same institute as Andreas Geiger & Philip Lenz when they created Kitti and they are close friends.

closed 2017-01-26 by SirVer

kentsommer, 2017-01-26:
Nice when its a one line fix. Thank you for such a detailed response @SirVer !

JoestarK, 2017-04-06:
@kentsommer 
Hello, kentsommer,
I also want to test cartographer on kitti dataset, but cartographer isn't publishing any scan matching or map using the fixed configure files you provided.

kentsommer, 2017-04-06:
@JoestarK 

I'm not sure what might be going wrong since the fix that SirVer gave did indeed allow things to run for me. However, I stopped working with the KITTI dataset shortly thereafter and have been using a private dataset since then, so, I have not been improving the configuration files provided since shortly after the fix was verified to fix my issue.

JoestarK, 2017-04-06:
@kentsommer 
Appreciate for your reply. I have solved this problem! Because the Topic of the Submaps is empty

davidswords, 2017-04-06:
Hi there,

As of yesterday, I was just having the same issue in relation to the map
frame being missing.
It seems to me that the cartographer_node doesn't publish that map
transform unless all of the topics are available right?

SirVer, 2017-04-06:
> It seems to me that the cartographer_node doesn't publish that map transform unless all of the topics are available right?

​That is correct. You tell Cartographer what data it should expect and it will wait for the data, time-sort all of it and process it in-order​. For this time-ordered merge-sorting to work, it needs to wait for all sensors to provide at least one packet. Only then can it run SLAM and only then can it provide you a transform from your tacking frame to map.