cartographer-project/cartographer_ros #249: Configuration issues - sudden shifts in localization/inconsistent mapping

sockhands, 2017-01-16:
Config, Bag, Launch and URDF
[https://drive.google.com/open?id=0B2PBRcDFH0eTbHBucHZtVjZHb3M](url)

Sensors:
Sick LMS 151 lidar
Xsens MTI 30 IMU

I'm having trouble finding an optimal configuration for Cartographer. I used the configuration provided above, mostly copied from the backpack_2d demo (changed to standard laser, not multi echo). In my early tests the localization went off track by several meters relatively quickly, and correspondingly the map accumulated error (skewing/ghosting).

Increasing min_score and global localization score in the sparse_pose_graph config (to 0.7 and 0.8 respectively) significantly improved the SLAM. However, the moment the robot moved a couple meters out from inside our warehouse and into an adjoining car park (suddenly with substantially less around for the lidar scans to register) the localization failed catastrophically. The robot transform seemed to accelerate rapidly away from the map in Rviz, even as the robot stopped moving. Shouldn't the absence of acceleration from the IMU constrain the robot's position and prevent this from happening?

Thanks in advance for your help!

sockhands, 2017-01-17:
I've since set the following:

SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e3
SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.55
SPARSE_POSE_GRAPH.constraint_builder.global_localization_min_score = 0.8

as well as setting the ceres scan matcher to use 4 threads instead of one. I've also done a hard iron calibration of the IMU and seen significant improvements to SLAM. There is still the issue of sudden, large jumps in localization on occasion - seems to also occur when rolling over sudden bumps, as well as when moving into wider, more sparse areas.

SirVer, 2017-01-17:
My guess would be that bumps show in huge acceleration in your IMU - which might confuse the SLAM. Can you try plotting the IMU accelerations around bumps?

We are in a week long hackathon, so it might be a while until we have time to look into this issue.

sockhands, 2017-01-17:
Thanks for the reply, sorry for the inconvenient timing. I won't be able to plot the IMU data until tomorrow, but will do so as soon as possible.

As for preventing the large jumps when the robot moves into wider, sparser areas - with less laser hits? It seems the scan matching is only constrained by the IMUs angular velocity, but not linear acceleration? 

Are there any parameters in the solver I should adjust to improve this issue, in the absence of wheel odometry?

sockhands, 2017-01-18:
A more illustrative (and smaller) bag file. The point at which the robot moves into open space and the localisation jumps around is obvious here. 
https://drive.google.com/open?id=0B2PBRcDFH0eTME5tVVRrZ29EbU0

SirVer, 2017-02-06:
Sorry for the long delay in getting back to you.

I think your IMU is really messing with you in this example. I did not investigate what its outputs are exactly, but turning it off I was able to generate a pretty reasonable map from the `bump_filtered.bag` (first picture) and also from `2017-01-16-17-27-13.bag` (second picture).

I quickly investigated the IMU data and so far I could not see anything that was obviously wrong. The linear acceleration is in the correct units and also contains gravity. It also has quite little noise. However occasionally there are spikes in the acceleration - but these might well be correct. Looking at the output of `rosrun rqt_plot rqt_plot /imu/linear_acceleration/x:y:z` might help you make progress there.

I would also considering if you want to SLAM at 5cm. Your laser is pretty coarse and you will have an easier time at a higher resolution.

My changes can be found in the last few commits of  https://github.com/SirVer/cartographer_ros/commits/user/sockhands.

I used this commandline: `roslaunch cartographer_ros carto_offline.launch bag_filename:=2017-01-16-17-27-13.bag`.

[image]

[image]

> SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e3

this should not have a huge impact. This is how we filter outliers.

> SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.55

The histograms printed while Cartographer is running shows that you have a ton of loop closures constraints already, so lowering the score (which will raise the number of constraints) is probably not too useful. I'd rather ramp it up.

> SPARSE_POSE_GRAPH.constraint_builder.global_localization_min_score = 0.8
This should have no effect without multi trajectory.

> And as for preventing the large jumps when the robot moves into wider, sparser areas with less laser hits? It seems the scan matching is only constrained by the IMUs angular velocity, but not linear acceleration?

Generally, when the scan matching is bad, nothing much will save you. You are right that the IMU will constrain the movement, but if the laser sees nothing, doubly integrating the IMU accelerations will give you a ton of noise and a not great result which will yield even worse scan matching.
Looking at `mapping_2d/local_trajectory_builder.cc` it seems you are right that we are only using the IMU to track Yaw in 2D. @wohe Was that intentional?

> Are there any parameters in the solver I should adjust to improve this issue, in the absence of wheel odometry?

I think since your laser is alright, you should be good with the settings I have right now and not bother with the IMU as much as possible. If you find another failure case we might need to revisit that. Closing this for now. Let me know if you require more help.

closed 2017-02-06 by SirVer

sockhands, 2017-02-07:
Thank you! This helps a lot. Really appreciate your time and the thorough response.

SirVer, 2017-02-07:
My absolute pleasure, but also part of my Job. :)

sockhands, 2017-02-14:
If we elect to not use the IMU, are we assuming the robot moves only on a flat plane? I'm assuming the IMU is necessary in any environment with bumps, ramps, slopes and the like. 

If the IMU data from my sensor was not obviously wrong, how is it that removing it has improved SLAM so significantly?

SirVer, 2017-02-14:
@sockhands 2D SLAM assumes a flat world always. If you have ramps in your environment it will not work with or without IMU. 

> If the IMU data from my sensor was not obviously wrong, how is it that removing it has improved SLAM so significantly?

It was wrong enough apparently. If you were driving on slopes this also screws up the re-projection into the ground plane. Disabling it turns this off (since no way of knowing the current orientation is there) and maybe your environment still looks "similar enough" for scan matching only.

jediofgever, 2018-03-09:
Hi @sockhands  I have the same setup with you, but I dont have configuration file for sick tim, can you provide your configuration file ?