cartographer-project/cartographer_ros #251: some issues about backpack_2d

carl1218, 2017-01-22:
Hi,i have some issues about backpack_2d
1,Can it show the robot walking path in the rviz?
2,How to deal with changes in the floor，For example,From the first floor to the second floor.

SirVer, 2017-01-23:
> 1,Can it show the robot walking path in the rviz?

Not a Cartographer question, but a quick google gave me this: http://answers.ros.org/question/209224/show-robot-trajectory-in-rviz-real-time/

> How to deal with changes in the floor，For example,From the first floor to the second floor.

These have to be separate maps. 2D assumes a flat world. Changing floors violates this assumption. You have to finish the first map before leaving the top floor and start a new floor once you reached a flat surface again.

closed 2017-01-23 by SirVer

carl1218, 2017-01-23:
@SirVer 
Thank you very much.
How can i used the finish_trajectory service to finished the map and start a new map?

SirVer, 2017-01-23:
On another console you should be able to

$ rosservice call /finish_trajectory foo

Which puts the assets into ~/.ros/foo* and starts a new trajectory in the current map. This is probably not what you want, since you changed floors and want a new map entirely.

So instead, you do the service call and restart the cartographer node. Alternatively you could add a RPC that does the equivalent of restarting the node, but keep the node running. If you do this please send a PR.