cartographer-project/cartographer_ros #254: 2D SLAM with Neato LIDAR

bliebman, 2017-01-25:
Hi,

I'm currently working on a project where are using a Neato (XV-11) LIDAR sensor for 2D SLAM with cartographer_ros. We have attempted to encode the output from the LIDAR into sensor_msgs/LaserScan messages as input to Cartographer, but the rviz output looks like scattered lines with no loop closure.

My question is, how would you recommend we encode the data into the messages? The LIDAR sensor provides serial output at 4 degrees of distance measurement at a time. Should we populate LaserScan "echoes" with the 4 distances/intensities and send 90 messages with each revolution of the sensor?

Thanks!!!

kentsommer, 2017-01-25:
@bliebman 

If you are using the laser standalone (USB) without the rest of the Neato robot I suggest you take a look at this driver (officially supported up to Indigo but shouldn't be too much work to port to kinetic if you need): http://wiki.ros.org/xv_11_laser_driver

tutorial on how to use the driver can be found here: http://wiki.ros.org/xv_11_laser_driver/Tutorials/Running%20the%20XV-11%20Node

However, if you are running the full Neato robot you are better off going with this driver which also supplies a laser scan topic: https://github.com/jmtatsch/neato_robot

possibly helpful tutorial for Neato with Indigo: http://servo.texterity.com/servo/201412/?folio=42&pg=42#pg42

SirVer, 2017-01-25:
It sounds to me like you are unsure about how to construct a laserfan from your neato data, which is not a Cartographer related problem. @kentsommer's reply contains every information I could provide either, so I am closing this.

If you believe this is related to Cartographer, please reopen and attach a few screenshots from rviz showing your lasers fans and Carto results.

closed 2017-01-25 by SirVer