cartographer-project/cartographer_ros #255: Cartographer_ros node died

nangong, 2017-01-25:
Hello, when I launched a cartographer node for mapping, an error appeared as follows:
```
F0125 15:26:29.805143 19325 pose_tracker.cc:167] Check failed: (orientation_.inverse() * Eigen::Vector3d::UnitZ()) .dot(gravity_direction_) > 0. (0 vs. 0) 
[FATAL] [1485329189.805294727]: F0125 15:26:29.000000 19325 pose_tracker.cc:167] Check failed: (orientation_.inverse() * Eigen::Vector3d::UnitZ()) .dot(gravity_direction_) > 0. (0 vs. 0) 
*** Check failure stack trace: ***
    @     0x7fd366291daa  (unknown)
    @     0x7fd366291ce4  (unknown)
    @     0x7fd3662916e6  (unknown)
    @     0x7fd366294687  (unknown)
    @           0x5c9c9a  cartographer::kalman_filter::ImuTracker::AddImuLinearAccelerationObservation()
    @           0x5cadd8  cartographer::kalman_filter::PoseTracker::AddImuLinearAccelerationObservation()
    @           0x59a6aa  cartographer::mapping_2d::LocalTrajectoryBuilder::AddImuData()
    @           0x4e6ad0  _ZNSt17_Function_handlerIFvlSt10unique_ptrIN16cartographer_ros10SensorDataESt14default_deleteIS2_EEEZNS1_12_GLOBAL__N_14Node10InitializeEvEUllS5_E4_E9_M_invokeERKSt9_Any_datalS5_
    @           0x503cbf  _ZNSt17_Function_handlerIFvSt10unique_ptrIN12cartographer7mapping14SensorCollatorIN16cartographer_ros10SensorDataEE5ValueESt14default_deleteIS7_EEEZNS6_13AddTrajectoryEiRKSt13unordered_setISsSt4hashISsESt8equal_toISsESaISsEESt8functionIFvlS0_IS5_S8_IS5_EEEEEUlSA_E_E9_M_invokeERKSt9_Any_dataSA_
    @           0x5043c1  cartographer::common::OrderedMultiQueue<>::Dispatch()
    @           0x511771  cartographer::common::OrderedMultiQueue<>::Add()
    @           0x511936  cartographer::mapping::SensorCollator<>::AddSensorData()
    @           0x50f88f  cartographer_ros::SensorDataProducer::AddLaserScanMessage()
    @           0x4e3f28  _ZN5boost6detail8function26void_function_obj_invoker1IZN16cartographer_ros12_GLOBAL__N_14Node10InitializeEvEUlRKNS_10shared_ptrIKN11sensor_msgs10LaserScan_ISaIvEEEEEE_vSE_E6invokeERNS1_15function_bufferESE_
    @           0x4f021b  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5027e8  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7fd3675fb6b5  (unknown)
    @     0x7fd3675b5107  (unknown)
    @     0x7fd3675b5c33  (unknown)
    @     0x7fd3675fe1e5  (unknown)
    @     0x7fd3675e5e0b  (unknown)
    @           0x4e91fe  cartographer_ros::(anonymous namespace)::Run()
    @           0x4e275d  main
    @     0x7fd3648faf45  (unknown)
    @           0x4e3877  (unknown)
    @              (nil)  (unknown)
[cartographer_node-1] process has died [pid 19325, exit code -6, cmd /home/administrator/catkin_ws/devel/lib/cartographer_ros/cartographer_node -configuration_directory /home/administrator/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files -configuration_basename demo_husky.lua imu:=costly_imu odom:=odometry/filtered __name:=cartographer_node __log:=/home/administrator/.ros/log/7c6a9ffc-e2c5-11e6-a31d-708bcd9da97c/cartographer_node-1.log].
log file: /home/administrator/.ros/log/7c6a9ffc-e2c5-11e6-a31d-708bcd9da97c/cartographer_node-1*.log
```

My configuration is ,
```
include "map_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  use_odometry_data = true,
  use_constant_odometry_variance = false,
  constant_odometry_translational_variance = 0.,
  constant_odometry_rotational_variance = 0.,
  use_horizontal_laser = true,
  use_horizontal_multi_echo_laser = false,
  horizontal_laser_min_range = 0.00999999977648,
  horizontal_laser_max_range = 50.,
  horizontal_laser_missing_echo_ray_length = 1.,
  num_lasers_3d = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
SPARSE_POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options
```

and the launch file is,

```
<launch>
  <param name="/use_sim_time" value="false" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename demo_husky.lua"
      output="screen">
    <remap from="imu" to="costly_imu" />
    <remap from="odom" to="odometry/filtered" />
  </node>

  <!-- <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_husky.rviz" /> -->
</launch>
```

SirVer, 2017-01-25:
Please also provide your bag.

SirVer, 2017-03-20:
Closing due to inactivity.

closed 2017-03-20 by SirVer