cartographer-project/cartographer_ros #256: Sparse laser scanner and cartographer

makbut, 2017-01-25:
Hello! I am running some simulations in V-Rep and I want to create the map of an environment using cartographer. In V-Rep there is a robot which publishes laser scanner data, imu data and odometer data to the appropriate ROS topics.  The problem is with the laser scanner. When I configure it being sparse and having a resolution of 1 point per degree cartographer crashes, whereas when I configure it having a resolution of 4 points per degree, it works!

The error message is:
`... logging to /home/lea/.ros/log/46105d78-e2db-11e6-9648-6c71d9f2819e/roslaunch-lea-All-Series-4006.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
started roslaunch server http://lea-All-Series:36902/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /use_sim_time: True

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
WARNING: Package name "ceres-solver" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits and underscores.
process[cartographer_node-1]: started with pid [4024]
log4cxx: Could not read configuration file [/home/lea/catkin_ws/rosconsole.config].
[ INFO][1485335024.661422331][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/carnode.lua' for 'carnode.lua'.
[ INFO][1485335024.661747847][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO][1485335024.661789159][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO][1485335024.661984156][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO][1485335024.662019403][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO][1485335024.662240520][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO][1485335024.662274757][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO][1485335024.662519173][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO][1485335024.662552439][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 configuration_file_resolver.cc:40] Found '/home/lea/catkin_ws/devel/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO][1485335024.690198204][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 submaps.cc:176] Added submap 1
[ INFO][1485335024.690300013][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:44.000000  4024 map_builder_bridge.cc:37] Added trajectory with ID '0'.
[ INFO][1485335025.040721535, 3571308.000000000][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:45.000000  4024 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '621391681077500000'.
[ INFO][1485335025.040793065, 3571308.000000000][/cartographer_node][ScopedRosLogSink::send]: I0125 10:03:45.000000  4024 local_trajectory_builder.cc:275] ImuTracker not yet initialized.
[ WARN][1485335025.321827353, 3571308.250000000][/cartographer_node][ScopedRosLogSink::send]: W0125 10:03:45.000000  4024 residual_block.cc:131] 

Error in evaluating the ResidualBlock.

There are two possible reasons. Either the CostFunction did not evaluate and fill all    
residual and jacobians that were requested or there was a non-finite value (nan/infinite)
generated during the or jacobian computation. 

Residual Block size: 1 parameter blocks x 207 residuals

For each parameter block, the value of the parameters are printed in the first column   
and the value of the jacobian under the corresponding residual. If a ParameterBlock was 
held constant then the corresponding jacobian is printed as 'Not Computed'. If an entry 
of the Jacobian/residual array was requested but was not written to by user code, it is 
indicated by 'Uninitialized'. This is an error. Residuals or Jacobian values evaluating 
to Inf or NaN is also an error.  

Residuals:             -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan         -nan 

Parameter Block 0, size: 3

        -nan |          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan          nan           [... cut ...]
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.ceres_solver_options.max_num_iterations=150
--TRAJECTORY_BUILDER_2D.adaptive_voxel_filter.min_num_points=2 
--SPARSE_POSE_GRAPH.constraint_builder.adaptive_voxel_filter.min_num_points=2
return options`

My working and not working bag files are:

http://www.filedropper.com/sparse1degree
http://www.filedropper.com/dense025degrees

SirVer, 2017-04-24:
This fell through the cracks and the bag files are no longer available. Sorry for this :(.

If this is still an issue, please open a new issue and reattach the data.

closed 2017-04-24 by SirVer

ZRazer, 2018-07-16:
Hi, have you ever mad any progress with this issue? I'm working with V-REP and cartographer, but still stuck in the same point.