cartographer-project/cartographer_ros #263: Dependencies problem with qtbase5-dev package

tsecoITA, 2017-02-10:
I followed the detailed instructions in https://google-cartographer-ros.readthedocs.io/en/latest in order to install cartographer_ros.
When the system tries to install all the needed dependencies with:
`rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y`
an error arrises:
```
The following packages have unmet dependencies:
 qtbase5-dev : Depends: libgles2-mesa-dev or
                        libgles2-dev but it is not installable 
```
I tried different ways to install libgles2-mesa-dev without exits.

I checked cartographer instalation in several computers and it seems to be a problem about the kernel version. In an Ubuntu 14.04 LTE with 4.2.0-30-generic there is no problem and all the dependencies are installed without any error (included libgles2-mesa-dev), but with 3.19.0-68-generic kernel version the dependencies problem arrises.

Without all the dependencies it is not possible to compile cartographer_ros package.
Do you have any issue similar to this ?
Thanks in advance.

SirVer, 2017-02-10:
Please check our Dockerfiles to see how we compile Cartographer. Our CI is not reporting any errors, so it works on the ubuntu versions we support.

tsecoITA, 2017-02-10:
Thank you for your quick reply. 
I followed the compilation instructions of your Dockerfiles.
The versions that you support is ubuntu 14.04 and 16.04 but, is there any known restriction about the kernel version?
Best regards,

SirVer, 2017-02-10:
No, we do not know about restrictions in kernel versions. However, what makes you believe that the OpenGL packages depend on the used kernel version? I would rather expect this to be only dependent on the OS version you are using.

tsecoITA, 2017-02-13:
We have tried to install cartographer_ros in several computers and the only difference among then is the kernel version, always with the same dependency error about libgles2-mesa-dev package.
With kernel versions 4.2.0-30-generic and 4.2.0-27-generic there is no problem but with 3.13.0-101-generic and 3.19.0-68-generic the problem persists.
For this reason I think that the kernel version has to do with the problem.

SirVer, 2017-02-13:
Interesting and thanks for clarifying.

Unfortunately, there is nothing Cartographer can do to fix this issue since we have no control over the kernel.

closed 2017-02-13 by SirVer

damonkohler, 2017-02-20:
This sounds similar to the issues described in the Indigo installation instructions: http://wiki.ros.org/indigo/Installation/Ubuntu

Perhaps the workarounds suggested there can help.

tsecoITA, 2017-02-21:
Thank you very much for your suggestion, but I tried the suggestions about the dependencies issue and it does not work. The complete information about the version of my ubuntu system is:

Linux ita866x 3.19.0-68-generic #76~14.04.1-Ubuntu SMP Fri Aug 12 11:46:25 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

I would like not to upgrade the kernel version because I am afraid that some code would stop working.
Any other suggestion would be grateful.

SirVer, 2017-02-21:
Run Cartographer in a docker container? Not sure if that works though, since docker probably shares the kernel version. You can also disable the qt RVIZ plugin (by commenting it out of the cMake file) and live without visualization.

narutojxl, 2018-11-23:
@tsecoITA , maybe this link can help you [https://github.com/googlecartographer/cartographer_ros/issues/1105](#1105)