cartographer-project/cartographer_ros #265: Misc: Cannot turn off Submaps rendering in Camera plugin view

ojura, 2017-02-16:
I am using the Camera plugin in rviz, which renders the 3D world as viewed from the camera coordinate frame superimposed on the camera picture. This plugin has a checkbox list of the objects that have been added in rviz, and their visibility in the camera view can be turned off by deselecting the respective checkboxes. When using the Submaps plugin from `cartographer_rviz`, a checkbox appears for it, however, the plugin does not respect this setting. The only way to remove the map from the Camera view is to remove or disable the Submaps plugin from the rviz session, however, in that case, you can't use it in the main view.

SirVer, 2017-04-24:
@ojura Is this fixed through your recent changes to the rViz plugin in cartographer_ros?

ojura, 2017-04-24:
No, this is not not related. See this video for the illustration of the bug: https://www.youtube.com/watch?v=D4gqg96Y6Zs&feature=youtu.be

Notice how the LaserScan visualization plugin respects the "Visibility" setting of the Camera plugin, while Cartographer's submaps plugin does not.

SirVer, 2017-04-24:
Thanks, that video made the use case quite clear. SubMaps do not seem very useful in the camera view though, so maybe they should just be permanently disabled.

Also, will you implement the necessary change? I think we do not care strongly about this, so it is unlikely that Cartographer team will pick this up.

ojura, 2017-04-24:
> Also, will you implement the necessary change?

I haven't debugged this properly (yet), so I am not sure what exactly needs to be changed in order to play well with the camera plugin.

> SubMaps do not seem very useful in the camera view though, so maybe they should just be permanently disabled.

When you set up transformations so the map is rendered on a floor plane -- and not very near the camera, which just causes the map to show up as Z fighting, like in my example -- it can look nice when the map is embedded in the video on the floor.

I think it won't make a large difference in coding effort whether to disable it permanently in the Camera view, or to respect the setting in the camera plugin - the biggest issue is to find out how to do it at all.